intelligent-agent/Refactor #7: Enhancement request: console screen rotation

manbehindthemadness, 2016-12-17:
As the console rotation is independent of toggle's orientation I suggest we change fbcon=rotate to 4 in /boot/uEnv.txt so the console output isn't locked into portrait format.

goeland86, 2016-12-17:
I will test this before including it in the release - at the moment I've noticed that on the original manga-screen the console is showing in landscape, so I wonder if it's different from screen to screen?

manbehindthemadness, 2016-12-17:
To branch this request I believe it would be prudent to sport the running journalctl -f output after bootup. This would provide easy diagnostics in the event a configuration problem causes toggle or networking to fail. If desired this can be achieved using this:

apt-get install rungetty
useradd -m dummy
echo "journalctl -f" >> /home/dummy/.profile
text='ExecStart=-/sbin/getty -a DUMMY 115200 %I'
sed -i "/.*ExecStart*./ c $b" /etc/systemd/system/getty.target.wants/getty@tty1.service

manbehindthemadness, 2016-12-17:
I am testing using a standard lcd monitor, so I think the manga screen might be unique, I wonder if there is a way we can auto switch it?

goeland86, 2016-12-17:
I'm gonna leave this one out for now - it's not a blocker, only a nice to have feature anyway. If you have an LCD chances are you want to run Toggle or some other UI on it, not a terminal... While helpful for debugging it's not a blocker either in my opinion. Thanks for bring it up, but this needs discussion with @eliasbakken I think.

goeland86, 2017-11-12:
@manbehindthemadness do you think this is still relevant to add? Assuming the config of the MS-2 is the same as MS-1, I'm wondering if this is still required? I mean, most guys who'll be debugging Toggle will be devs with SSH access, and if folks plug in a regular HDMI screen, it would then be harder for them to work with it. What do you say?