intelligent-agent/Refactor #8: Cura Engine

FuraX49, 2016-12-17:
Version 15.04.06 is no longer available on ultimaker repository...

goeland86, 2016-12-17:
It is, but you need to look at the tags for 15.04.6, not .06. See https://github.com/Ultimaker/CuraEngine/tree/15.04.6

FuraX49, 2016-12-17:
I just test your script , and i see error...
I verify with  bash 👎 
git checkout tags/15.04.6
error: pathspec 'tags/15.04.6' did not match any file(s) known to git.

goeland86, 2016-12-17:
Ok checkout again - the reason the checkout failed is because I cloned with depth 1.
I switched to pulling the zip from the github repo instead. The idea is to save as much disk space as possible.

FuraX49, 2016-12-17:
86 as Vienne ?

goeland86, 2016-12-17:
try a git pull - you'll see the changes I made to the install_cura method.

FuraX49, 2016-12-17:
I try to reclone the repository...

FuraX49, 2016-12-17:
After recloning repository it's works....
Sorry

closed 2016-12-17 by FuraX49