intelligent-agent/Refactor #10: Network manager constantly calling wlan0

manbehindthemadness, 2016-12-18:
Network manager is calling wlan0 for signal strength, even if it's not present. Journalctl is overflowing with the following:

 <warn>  [1482078765.9286] (wlan0): error getting signal strength: No such device

manbehindthemadness, 2016-12-18:
Resolution:
http://askubuntu.com/questions/764510/networkmanager757-warn-1461836136-9728-wlan0-error-getting-signal-stre

In file /etc/NetworkManager/NetworkManager.conf under [main] change:

dhcp=dhclient

to

dhcp=internal

goeland86, 2016-12-20:
If I read this right, in the end we're not just changing the file, you had to recompile a complete network-manager package, right?

goeland86, 2016-12-20:
I'm marking it as an enhancement as it really isn't a blocker, just a minor annoyance.

closed 2017-01-17 by goeland86