intelligent-agent/Refactor #12: enhancement request: Public samba share to uploads folder[REVISED]

manbehindthemadness, 2016-12-18:
apt-get install samba
cat <<EOT >> /etc/samba/smb.conf
dns proxy = no
log file = /var/log/samba/log.%m
syslog = 0
panic action = /usr/share/samba/panic-action %d
server role = standalone server
passdb backend = tdbsam
obey pam restrictions = yes
unix password sync = yes
passwd program = /usr/bin/passwd %u
passwd chat = *Enter\snew\s*\spassword:* %n\n *Retype\snew\s*\spassword:* %n\n *password\supdated\ssuccessfully* .
pam password change = yes
map to guest = bad user
usershare allow guests = yes

[homes]
   comment = Home Directories
   browseable = no
   read only = no
   create mask = 0777
   directory mask = 0777
   valid users = %S

[printers]
   comment = All Printers
   browseable = no
   path = /var/spool/samba
   printable = yes
   guest ok = no
   read only = yes
   create mask = 0700

[print$]
   comment = Printer Drivers
   path = /var/lib/samba/printers
   browseable = yes
   read only = yes
   guest ok = no
   
[public]
   path = /usr/share/models
   public = yes
   writable = yes
   comment = smb share
   printable = no
   guest ok = yes
   locking = no
EOT

goeland86, 2016-12-20:
You committed this to the dev branch right?

goeland86, 2016-12-20:
Yep I see it in there and it's now done.

closed 2016-12-20 by goeland86