intelligent-agent/Refactor #20: BBB Wireless not supported

gruvin, 2017-01-07:
My first BBB happens to be the BBB Wireless. 

Using Kamikaze-2.1.0-rc1.img.xz, the boot fails as follows ...

```
...
...
Running uname_boot ...
loading /boot/vmlinuz-4.1.36-bone24 ...
6921888 bytes read in 467 ms (14.1 MiB/s)

unable to find [dtb=am335x-boneblack-wireless.dtb] did you name it correctly? ...

FAILSAFE: U-Boot UMS (USB Mass Storage) enabled, media now available over the usb slave port ...
UMS: LUN 0, dev 1, hwpart 0, sector 0x0, count 0x720000
musb-hdrc: peripheral reset irq lost!
```

I think its the `wl81xx` chipset, like on some BBB WiFi cape I noticed, somewhere. I'm too knew to all this to be certain.

I tried a manual build of Kamikaze2.1, starting from a working (WiFi connected) Debian image. But after completion of `bash make-kamikaze-2.1.sh` the result was the same as above.

Seems I cannot build the TI `wl81xx` drivers manually, because I cannot `apt-get` the requisite tools, to begin with. Chicken and egg scenario. Doesn't seem that would help anyway, since the Kamikaze script appears to overwrite the kernel, anyway. Right?

Well, thanks for looking into this. I'll put my $hiny new Replicape in a drawer until later, I suppose. :-P

goeland86, 2017-01-07:
Hi Bryan, 

Thanks for trying it out on the wireless chipset, I haven't been able to try it. The dtb file comes from the overlay I think, and I'll have to look into how to build kamikaze with it (or for it).

gruvin, 2017-01-07:
Thanks for that, @goeland86.

I'd like to learn more about this, "overlay" thing and just in general how the whole BB kernel/software fits together in the first place. Guess I'll get Googling. ;-)

gruvin, 2017-01-07:
I'm attempting a Kamikaze install, without up(down)grading to kernel 4.1, by commenting out that line in `prep_ubuntu.sh`.  This should leave my system at kernel 4.4 and thus, presumably, allow networking to continue working. 

From there, I assume Kamikaze will need to compile some C drivers and I'm hopeful that will still work with kernel 4.4. 

No doubt, you already know this will or will not work. But I'm trying it anyway. :-P

If it doesn't work, then I've discovered that the dtb file thing _might_ be as simple as copying the file into `/boot/dtbs/<kernel_version>` and maybe editing `uEnv.txt` to use either that or the right one. Haven't got that far, yet. 

I'll let you know.

goeland86, 2017-01-07:
That approach of copying the dtb file might be the better one, but I can't
say for sure yet. I'll see if I can figure it out better, but I think that
is the better approach. Redeem was having issues with 4.4 so that was the
why of downgrading to 4.1... Keep posting what you've found out about it
here, it will help a lot of people I think.

gruvin, 2017-01-07:
Noted about 4.4. Thanks.

Interestingly, the `apt-get upgrade` (in `prep-ubuntu.sh`) resulted in removing my `/boot/dtbs/4.4.xxxx/` folder, replacing it with `4.8.7-arm7-x4/`. I would have expected an additional folder, not a replacement. *shrug* In any case, my system still boots to kernel 4.4!

So, it turns out that `uEnv.txt` has all the dtb file loads line commented out. That explains why everything _still works_. This makes me wonder if the dtb file for kernel 4.1 is actually required, in the first place.

Anyway, just for kicks, I'm currently running `make-kamikaze-2.1.sh` on my kernel 4.4 system. I figured I'd at least be able to watch the build process, before potentially bricking the thing again.

When this is done, I'll attempt a downgrade to kernel 4.1, using the tool script referenced in `prep-ubuntu.sh` and see about resurrecting the broken system from there. Assuming that works (based on my assumption that kernel drivers get compiled) I'll do a clean build of Kamikaze again.

More to come.

gruvin, 2017-01-08:
Interesting. After the Kamikaze build, the external MMC card no longer boots ...

```
U-Boot SPL 2016.09-rc2-dirty (Sep 14 2016 - 17:44:40)
Trying to boot from MMC1
Expected Linux image is not found. Trying to start U-boot
```

So, it turns out that `/boot/config-4.4.30-ti-r66` has now been deleted and replaced by `config-4.8.7-armv7-x4`. The why of this, I cannot fathom. I just changed `uname_r=4.4.30-ti-r66` in uEnv.txt to `uname_r=4.8.7-armv7-x4`. 

So, now I'm booting into k4.8, whether I like it or not. (Kernel 4.8 was installed by `apt-get upgrade`, during execution of `prep-ubuntu.sh`, in an earlier step. But the system was still booting k4.4. I saw no reason to change it. But now the config file has been blown away. {shrug}

... oh and, now WiFi is broken again ...

```
[  246.629939] wlcore: ERROR Your WiFi FW version (8.9.0.0.39) is invalid.
[  246.629939] Please use at least FW 8.9.*.*.58.
[  246.629939] You can get the latest firmwares at:
[  246.629939] git://git.ti.com/wilink8-wlan/wl18xx_fw.git
```

So much for kernel 4.8.

- - -

I could continue down this road, I guess. However, given that I've been told that Redeem has issues with kernel 4.4 let alone k4.8, that doesn't seem to make much sense. 

So instead, I'll stop wasting time and return to the offical Kamikaze2 image, to set about fixing the dtb file issue. 

Guess I'll take a `dd` image backup of this card first, just in case I want to return to it, later.

goeland86, 2017-01-08:
Yeah, that's unfortunate...

The kernel update scripts are not just using apt get, and someone pointed
out on slack that the latest kernel patch for the wireless BBB dates from
Dec 26, so it may not have propagated through all the way, hence the wifi
errors you see.

Good luck though, I hope you find a way to make it work!

gruvin, 2017-01-08:
OK. So, I used `dtb-builder` to obtain a `am335x-boneblack-wireless.dtb` file for kernel 4.1.x and installed that in the Kamikaze2.1RC1 image's `/boot/dtbs/...` folder. This allowed the kernel to boot without the, "`unable to find [dtb=am335x-boneblack-wireless.dtb] ...`" error.

But now the kernel cannot seem to find any MMC storage devices ...

```
(initramfs) ls /dev/mmc*
ls: /dev/mmc*: No such file or directory
```

I also tried to boot using the already existing `am335x-boneblack-wl1835mod.dtb` file, since that seems to best match the labelling on the WiFi/BT module itself. With that, the kernel loads and runs fine. But things fail at around the time network interfaces are trying to start. The start-up script gets stuck in a loop, forever. (My stupid terminal software doesn't allow me to see the messages correctly, due to some scrolling bug / wrong-emmulation thing.)
 
Having so little knowledge of Linux on ARM and Beaglebone's in general, I believe I've now gone as far as I can. Sorry I wasn't able to crack it for y'all.

gruvin, 2017-01-08:
I found an old USB-to-Ethernet dongle in my junk drawer. This allows me to bypass the WiFi issues on the BBB Wireless, for now. 

Meanwhile, I'm going back to ground zero, starting with installing Redeem manually, atop of kernel 4.4, from [this 2017-01-04 Debian image](
https://rcn-ee.net/rootfs/bb.org/testing/2017-01-04/). Running `apt update / upgrade` right out of the gate appears to have updated the cape overlays from v4.1 to 4.4.

I'm concerned about the previous comment re Redeem having issues with k4.4. But I've crossed my fingers ... so, everything should be just fine. :-P 

Maybe going this route will at least get me to a point where I can play with the Replicape.

gruvin, 2017-01-09:
Hmmm ... 

```
slot #4: BB-SPIDEV1 conflict P9.31 (#0:BB-BONE-REPLICAP)
```

And yet ...

```
root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: P---L-   0 Replicape 3D printer cape,0B3A,Intelligent Agen,BB-BONE-REPLICAP
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
```

and

```
##BeagleBone Black: HDMI (Audio/Video)/eMMC disabled:
dtb=am335x-boneblack-overlay.dtb
```

Should I just give up? :-/

gruvin, 2017-01-09:
I finally got a working system on the BBB Wireless, using an external USB Ethernet adapter and foregoing WiFi entirely, for now.

The first and main issue was not being able to boot the **Kamikaze2 image** on the BBB Wireless. The complaint was a missing `.dtb` file. I fixed that by editing `uEnv.txt` and uncommenting the following line ...

```
dtb=am335x-boneblack-overlay.dtb
```

Now the kernel boots up. But because there's no `eth0` phy, the start-up scripts block. The fix for that was to comment out the following two lines in `/etc/network/interfaces` ...
 
```
#auto eth0
#iface eth0 inet dhcp
```

I found an old Apple USB-to-Ethernet adapter in my junk drawer and used that to get Internet connectivity. Using `dmesg`, I found that it had some funky device name, resulting in the addition of the following, again in `/etc/network/interfaces`) ...

```
auto enxb88d1254404f
iface enxb88d1254404f inet dhcp
```

There was a really weird, funky thing going on with the serial console, making it completely unusable. The fix for that turned out to be disabling the USB network bridge. In `/etc/network/interfaces` ...

```
#iface usb0 inet static
#    address 192.168.7.2
#    netmask 255.255.255.252
#    network 192.168.7.0
#    gateway 192.168.7.1
```

So now, finally, visiting `http://kamikaze.local` (which take a long time to load, the first time) eventually results in OctoPrint's start-up page. 

Looking in `/dev/` I see ...

```
root@kamikaze:~# ls -1 /dev/octoprint_*
/dev/octoprint_0
/dev/octoprint_1
```

Not sure why there are two. But OctoPrint only shows the `_1`. So whatever.

There is no `wlan0` interface, despite using the `am335x-boneblack-overlay.dtb` file, which suggests to me that is _should_ be present. So, still no WiFi. :'( 

At least I can play with Redeem now, to get to know how it works and so on.

In hindsight, a standard BBB Ethernet board, with an external USB WiFi dongle, would have been a better idea! Live and learn.

Hope that helps.

goeland86, 2017-01-09:
OK good news that! As to why the USB dongle wouldn't have worked, yeah the original USB Ethernet module would have interfered with it. 

Now, looking at the onboard wifi, the problem is that the overlay for Linux of that board is way too recent. The patch for the kernel was pushed Dec 26, so it hasn't made its way into the Ubuntu kernels yet. We'll also probably need a different dtb file for it but I'm not positive about that yet. 

I'll try to get my hands on a wireless board to keep development going with it

gruvin, 2017-01-09:
Awesome. Thanks.

gruvin, 2017-01-10:
Thanks again for your help so far, @goeland86. I confess, I'm a little confused as to which repo I should be hitting up for further assistance. I originally thought this fork was the original project, for some reason, you see! ;-)

Could you please take a look at my [latest comment over here](https://github.com/eliasbakken/Kamikaze2/issues/8#issuecomment-271449362) and advise the better place, going forward? Thanks a bunch.

goeland86, 2017-01-10:
No worries, I took over to push 2.1 because @eliasbakken is busy building the new mangascreeen design for production. I'll have a look, but I am only flying home from holidays tonight, so no promises as to being able to help quickly.

gruvin, 2017-01-10:
Understood and thanks again. Hope you had/have a good flight and come home to everything in order. :-)

There's no rush here. Really appreciate your time, just whenever it's convenient.

gruvin, 2017-01-10:
OK. Overlays now mostly understood. In particular, the fact that we can now load/unload them from user land. Nice! I've also figured out that Kamikaze disables the Cape Manger, which was confusing me before.

What I don't yet understand is how or when the `BB-BONE-REPLICAP_*` overlay is loaded. I see no reference to it in any of Redeem's source code (except the I2C EEPROM key writing. But that's nothing to do with the Replicape overlay, as such). So, am I right to guess, then, that the `am335x-boneblack-overlay.dtb` overlay I kinda randomly chose in `/boot/uEnv.txt`, just happens to include the `BB-BONE-REPLICAP_*` dts data? That seems reasonable, given that I'm using the Kamikaze image, after all.

Obviously, I hope, I'm in absolutely no rush for any of this. Everything is working at present. I'm just being a curious hacker. ;-)

m0n5t3r, 2017-01-12:
I, for one, am still searching for an image, any image, where wifi works on this thing :) (with the latest builds from https://rcn-ee.net/rootfs/bb.org/testing/ I get "wlcore: ERROR ELP wakeup timeout!"), and kamikaze has the issues above

luckily I'm not in any hurry either, I am waiting for some hardware for a delta printer and also have a normal BBB to use with it :)

gruvin, 2017-01-12:
@m0n5t3r Turnes out that those are not in fact, the latest images. Try [here](https://rcn-ee.net/rootfs/bb.org/testing/2017-01-04/). Anything before then (late 2016) didn't work for me.

m0n5t3r, 2017-01-13:
lol, so wifi works only when flashed to the emmc and if I select the beaglebone-wireless dtb explicitly; for some reason; going to try getting kamikaze to work on the thing during the week-end then, as soon as I understand what the hell a DTB is, and if it's transferable between kernel versions

goeland86, 2017-01-13:
So. The DTB file is a linux overlay. Basically to avoid having multiple dozens of ARM architectures in the linux kernel they built the overlay system allowing for a more generic kernel code, and the overlay binaries giving the necessary specifics to the kernel at run-time (if I got my story straight). So the reason wifi only works with the beaglebone-wireless dtb is exactly that it's the DTB telling the kernel about the onboard wifi chip, which the normal BBB DTB tells it about the Ethernet port which doesn't exist.

m0n5t3r, 2017-01-13:
I thought they were auto-detected, silly me :D (I saw on a forum something something eeprom id)

so, I took the SD card with kamikaze, copied over am335x-boneblack-wireless.dtb from the debian image I had working (2017-01-04, but it should work with the latest as well methinks), added `dtb=am335x-boneblack-wireless.dtb` to `boot/uEnv.txt` and I've got wireless with the "official" kamikaze image, running from the SD, even :)

so, supporting this might be easier than expected, it's enough if the next build includes the appropriate dtb file and a note tells wireless users to do add the line to uEnv.txt

GearheadRed, 2017-01-13:
I dunno if you guys already figured this out or not, I have only been loosely paying attention to the email thread here(distracted with a newly working printer now). It looks like the more recent BBB-eMMC flasher image has the wireless DTB files in them:
/boot/dtbs/4.9.3-armv7-x4/am335x-boneblack-wireless.dtb
/boot/dtbs/4.9.3-armv7-x4/am335x-boneblack-wireless-emmc-overlay.dtb

Location found from: https://rcn-ee.com/rootfs/2017-01-12/flasher/BBB-eMMC-flasher-ubuntu-16.04.1-console-armhf-2017-01-12-2gb.img.xz

I am working on flashing this to my wireless BBB currently to try and confirm that its working. If it does being the updated base image I will likely give building kami on top of it a go as well.

m0n5t3r, 2017-01-13:
mkay, so I had Kamikaze 2.0.8 (debian) on that SD card; eventually I managed to get Kamikaze 2.1 almost working (meaning, everything works apart from ... redeem)

* booted with kernel 4.8, which is missing modules
* got networking up via usb (I just set my laptop as the default gateway and made it route traffic from the beaglebone)
* tried a bunch of kernels, discovered that wifi works with pretty much every kernel newer than 4.1 if I download the newest firmware from TI 
* fumbled around the internets looking for `pvrsrvkm` and `omaplfb`, which are apparently needed for toggle; eventually came across `/usr/srv/Kamikaze2`, found out I need `ti-sgx-es8-modules`, chose `ti-sgx-es8-modules-4.4.40-bone16` and the corresponding kernel; I needed to run `depmod -a` manually, the package post-install doesn't run it automatically for some reason

Now redeem doesn't start because systemd is waiting for an spi device that isn't there, even with the replicape connected - it is seen by the kernel:

```
[    4.228535] bone_capemgr bone_capemgr: slot #0: 'Replicape 3D printer cape,0B3A,Intelligent Agen,BB-BONE-REPLICAP'
```

but 
```
root@kamikaze:~# ls -l /sys/devices/platform/ocp/481a0000.spi/spi_master/
total 0
drwxr-xr-x 5 root root 0 Jan 13 22:43 spi1
```

and systemd is waiting for `sys-devices-platform-ocp-481a0000.spi-spi_master-spi2-spi2.1-spidev-spidev2.1.device` and times out eventually

m0n5t3r, 2017-01-14:
this explains why it was failing to detect it's a beaglebone wireless and was selecting the generic dtb, looks like the recent batch sold by mouser has a misconfigured EEPROM: https://groups.google.com/forum/?hl=en#!category-topic/beagleboard/beaglebone-black-wireless/VftTsmNFIUc

goeland86, 2017-01-14:
Ouch, ok.

Well, let me summarize if I can, because I'd like to be able to use this issue to track the development of Kamikaze to support the Wireless BBB:

- Wifi works if using the BBB wireless DTB with a recent kernel. @m0n5t3r can you confirm 4.8.x or 4.9.x kernel version is ok?
 - You only need to modify the uboot line to alternate the DTB between the ethernet and wifi BBB versions, which means if we have a functioning kamikaze 2.1 with a recent kernel, we can add both DTB files and let BBB wireless users edit the file before flashing the SD to the eMMC

Thus the only real problem is getting redeem to function on a modern kernel, correct?

m0n5t3r, 2017-01-14:
Wifi: I tested with 4.8 and 4.9, they worked; the version of 4.8 that is included (leftover?) in the latest kamikaze 2.1 image is missing /lib/modules, though, you need to reinstall it. 

It seems you can use any version newer than 4.1 that has a wireless dtb (4.1 fails to detect mmc and drops you into the initramfs shell eventually), as long as you get the firmware from `git://git.ti.com/wilink8-wlan/wl18xx_fw.git` and drop `wl18xx-fw-4.bin` in `/lib/firmware/ti-connectivity/`; the latest commit in that git tree is from August 2016, but it seems linux-firmware hasn't caught up yet in Ubuntu (Debian has a newer version); it looks like all 4.4 and later kernels also come with the necessary dtb for wireless, so the only change needed is to add the dtb line to uEnv.

However, you need a kernel that has a `ti-sgx-es8-modules` package if you want toggle; this restricts you to 4.0 to 4.4 series kernels, unless you enjoy compiling and apparently registering on TI's site to download the SDK, etc; having spent most of the late 2000s in Gentoo, I think I've had enough kernel compilations on slow hardware to last me for a long time :D

So yes, now the only issue remaining is that missing spi2 bus, which prevents redeem from working.

m0n5t3r, 2017-01-14:
Further investigation: it seems that SPI1 device works; I changed ShiftRegister.py a little to do spi.open(1,1) if spi.open(2,1) gives an IOError, started redeem in a console, connected a stepper, gave it some G1 commands from octoprint and it spins as expected :)

So, how can we rename SPI1 to SPI2? :)

GearheadRed, 2017-01-17:
I put some work in referencing the work from others here and tried to build out a functional wifi compatible image based on the 4.4 kernel. The dev branch of this git has the updates to the scripts if anyone wants to reference them.
Heres the image I came up with for wifi support and kernel 4.4 https://drive.google.com/open?id=0B9nRYtGq5muNVHpZVEZmS284anM I need to test it more when I get home with a replicape attached since I cant check for the SPI channel issues noted. Everything seems to be working as expected up to that point so far. It was noted that the issues for the SPI stuff were worked around within Redeem via slack convo.
This gDrive url will eventually become invalid, this is my personal gDrive so I will likely remove this image at some point in the future. The image is 2.2 gigs in size uncompressed so you will need at least a 4GB SD card to write it to.

GearheadRed, 2017-01-17:
I also wrote this wiki, you can replicate my work using the instructions there https://github.com/goeland86/Kamikaze2/wiki/Kamikaze-Image-Create-Logic

Only point to notice is that the instructions default to using the master branch. You will want to switch to the dev branch after cloning in the "Clone the Kamikaze2 git repo:" under the "Prepare Base Image" section if you are trying to follow along with building an image for the wifi boards:
```
cd /usr/src && git clone http://github.com/goeland86/Kamikaze2
git checkout dev
git pull
```

GearheadRed, 2017-01-18:
OK, SPI was still broken in that image. I was not at home at the time to test with an actual replicape. I am working on building a new image now after adding the SPI changes to git and confirming its working.

GearheadRed, 2017-01-18:
OK, Image was updated to fix the SPI issue. The same link as prior will work to download the updated image, prior image has been replaced. https://drive.google.com/open?id=0B9nRYtGq5muNVHpZVEZmS284anM

gruvin, 2017-01-18:
Great news! Thanks!

GearheadRed, 2017-01-23:
OK, New image has been uploaded. [Same URL as prior.](https://drive.google.com/open?id=0B9nRYtGq5muNVHpZVEZmS284anM) This image should encompass pretty much all fixes at this point, if someone can test it and confirm its working for them in the same way its working for myself it will likely end up being Kamikaze 2.1.1rc1.  This image has Ubuntu 16.04, kernel 4.4, and Octoprint 1.3.1rc2 (theres a bug currently causing issues with the setup wizard on the 1.3.0 tagged release so I bumped up the version to work around it).

gruvin, 2017-01-26:
@GearheadRed ... Just to let you know that this is very much appreciated. The time gods haven't allowed me a chance to test it yet though. I will let you know, just as soon as I have.

m0n5t3r, 2017-01-26:
managed to boot it now, seems fine (as in, everything starts apart from the bluetooth thingy, which is not critical for system functioning)

Octoprint 3.1 just got released, btw.

GearheadRed, 2017-01-27:
I am not sure about the bluetooth part, I have not put any effort into trying to make sure bluetooth is working properly, likely would be good as a separate issue to tackle once wifi part of wireless is working.

[Download from same URL as prior](https://drive.google.com/open?id=0B9nRYtGq5muNVHpZVEZmS284anM)

Rough outline of changes:

I have rebuilt the image with some changes to what package is installed for network manager and how the setup script goes about it currently to get the newer network manager version.  This method so far has tested to resolve the occasional issues with dns/resolvers not working even with dhcp prior. In coming months this will go away as Ubuntu releases 16.04.1 and the newer network manager version is available.

I switched to the OctoPrint 1.3.1 release tag(update page still complains about commit tracking instead of release tracking, need to look into that more).

Added power commands to OctoPrint config file to go along with preconfigured Webcam settings.

Proper references to release 2.1.1 were added to OctoPrint.

Switched to dedicated thing-printer Ubuntu repo instead of Debian.

goeland86, 2017-02-04:
Can we close the issue and put any further images as 2.1.1 RC images? Or is that premature, @GearheadRed ?

gruvin, 2017-02-04:
I haven't yet tested this, sorry. Just haven't had the chance. But I'm happy to run with whatevet @GearheahRed suggests. Thanks.

goeland86, 2017-02-04:
We're entering officlal RC testing, so I'm closing the issue.

closed 2017-02-04 by goeland86