intelligent-agent/Refactor #21: Octoprint wants to install an update at every boot

goeland86, 2017-01-18:
jetguy [reports](https://replicape.slack.com/archives/development/p1484745589004535) that Octoprint wants to install an update on every boot.

This means that somehow the update fails to install for some reason, or that the git permissions on the checkout folder prevent it from downloading the update as expected. Needs to be fixed.

jetguy, 2017-01-18:
[image]
[image]

jetguy, 2017-01-18:
It's more an issue of something right at the end with the restart. Also, actually using the restart octoprint works a
[image]
s in it reboots, but it throws an error saying it didn't?? Again, it definitely restarts octoprint and I had to capture that message before it through up the reconnect banner on the page.

jetguy, 2017-01-18:
I finally got a capture of the error details here.
[image]

goeland86, 2017-01-21:
This was fixed by addressing the installation method of octoprint (i.e. cloning the git isn't the proper way to do so - pip or easy_install is a better way of setting up the initial installation).

closed 2017-01-21 by goeland86