intelligent-agent/Refactor #27: Kamikaze 2.1.0 steppers only initialized after redeem startup

goeland86, 2017-02-08:
During power-on, Kamikaze 2.0.8 had a specific patch in uboot to quiet the motors as soon as uboot finished initializing. This is not functioning on 2.1.0 probably because of the switch to an Ubuntu base.

Needs fixing.

bliedblad, 2017-02-12:
Hi, I'm just bringing up a new printer with a Replicape I bought a few months ago. 
I used the 2.0.8 image to flash the BBB and ran some burn-in Gcode for XY motion from OctoPrint... it was super easy to get going from the point of flashing - thanks for the hard work!

I figured I would update to 2.1.0b and get wifi going before tacking fine tuning, but noticed his report on potential motor noise on boot. As I never ran anything but 2.0.8, I have never heard the noise. 

Questions:
1) how severe is the noise?

2) is this just an issue with the kamikazi build, or something due to differences between Debian and Ubuntu?

3) can this be fixed by just adding the patch into uboot?

4) as I don't have all my configuration worked out, and I'm ready to flash anyway - do you need an extra tester? What can I do to help?

goeland86, 2017-02-12:
Hi @bliedblad,

The noise is basically the initial stepper noise when the drivers are still in an uninitialized state. This happens only on the B3 boards - the B3A fixed that at the HW level. So if you have a B3A it's not going to affect your board.

For how to fix it, it's a patch to the uboot that needs to be done. Specifically for 2.0.8 @eliasbakken had built a new uboot for the image specific to the kernel it was running. 2.1.0 is running a different kernel and I think that's why the uboot fix isn't effective in 2.1.0. It should be fixed in 2.1.1.

As for extra testers - we always like to have more, but for now the development's progressing a little slow on my side as I'm going through a relatively busy patch in the year. I hope to have more time come end of march. I or @GearheadRed will put out RC images from time to time, and we appreciate any feedback/bugreports when those come out. We mostly discuss the fixes and enhancements for the images in the replicape slack's #development channel. Join us there if you want to be kept abreast of the news regarding kamikaze :)

Cheers,
Jon

pistolero, 2018-02-23:
@goeland86 do you have any details on those uboot fixes? Do you know why they are not working in 4.4 kernel?

goeland86, 2018-05-07:
@pistolero sorry for the long quiet - this issue dropped from what I was paying attention to.

Sadly, no, I do not know what the fixes were. I suspect it's simply a matter of doing a low pin assignment somewhere using a bash script as soon as the kernel is loaded, but I've no idea and not looked at how to fix this in newer builds...

eliasbakken, 2021-09-03:
I don't have the right cape to fix this issues I think, so I will close it without fix. If someone wants to take a look at this again later, I can assist.

closed 2021-09-03 by eliasbakken