intelligent-agent/Refactor #34: BBBW support on 2.1.0b

pomeroyb, 2017-03-31:
I see that support for the BeagleBone Black Wireless is "In Progress" according to the Projects tab. I'm looking to use the bbbw as my main board, and flashing 2.1.0b to it doesn't work at all. (All four LEDs light up solid and hang indefinitely).

How can I help make this happen? Do we know what the underlying issue is?

goeland86, 2017-03-31:
The issue has to do with the kernel overlay and kernel drivers to support the wireless chip,  that is different between the boards.

If you want to help, there's the 2.1.1rc1 image that's supposed to work on bbbw, if you can already test that it would be good feedback.

Another thing is if you were to join the development channel on the replicape slack. It becomes easier to coordinate what testing out development we're doing.

I still need to write contributing guidelines, but it's been a 3 person effort so far. Thanks for taking the time to help.

pomeroyb, 2017-03-31:
2.1.1rc1 worked for me! Only took about 5 minutes to go from the Cylon LEDs to boot.

Are there any log files you'd like me to upload?

I'd be happy to pop into the slack channel -- let me know what you need from me!

goeland86, 2017-03-31:
Glad to hear you got it working!

As for log files - I think that needs to be between you and @GearheadRed for the moment - I'm not in possession of a BBBW (yet) and he took the lead on the support for it.

The next question is whether or not 2.1.1rc1 is something you can print with. The main difference we're aiming at between 2.1.0 and 2.1.1 is BBBW support. And especially reliable wifi support - there are several reports of wifi being unreliable on BBBW. If you have any experience with linux dev/admin your help would be appreciated trying to debug that, as the amount of time @GearheadRed or myself can devote to fixing that goes.

pomeroyb, 2017-03-31:
I found a couple of log files waiting for me when I booted up, so here those are:

[prep_ubuntu.txt](https://github.com/goeland86/Umikaze2/files/884734/prep_ubuntu.txt)
[make-kamikaze.txt](https://github.com/goeland86/Umikaze2/files/884736/make-kamikaze.txt)

I'm pretty available to help debug, just let me know what you're looking for! :)

goeland86, 2017-05-16:
So I've managed to get a BBB-W myself for development and I'm running into kernel upgrades bricking the image... Need to figure out why it isn't booting afterwards. I'll post an image soon (I hope, been plagued with things to resolve now that weren't related to Umikaze). I'll ask for beta-testers then :)

goeland86, 2017-08-06:
Just FYI the Umikaze 2.1.1 RC3 works on both BBB and BBB-W :) I'm hoping it'll work for you as well?

closed 2017-08-06 by goeland86