intelligent-agent/Refactor #36: Trying to roll my own... connman removal issue?

bliedblad, 2017-04-05:
I'm attempting to create a new Kamikaze/Umikaze image using the directions found on the wiki home page: https://github.com/goeland86/Umikaze2/wiki

At first, I tried using the most recent image from http://elinux.org/BeagleBoardUbuntu (bone-ubuntu-16.04.2-console-armhf-2017-03-09-2gb), but keep running into an issue with the prepare ubuntu script. 
Seems like my terminal disconnects during the "install network-manager" step (fails when connman is removed).

If I follow the directions exactly (using the 16.04.1 build), I get a few points where I'm prompted for input:

> Configuration file '/etc/issue'
>  ==> Modified (by you or by a script) since installation.
>  ==> Package distributor has shipped an updated version.
>    What would you like to do about it ?  Your options are:
>     Y or I  : install the package maintainer's version
>     N or O  : keep your currently-installed version
>       D     : show the differences between the versions
>       Z     : start a shell to examine the situation
>  The default action is to keep your current version.
> *** issue (Y/I/N/O/D/Z) [default=N] ? 

> 
> Configuration file '/etc/issue.net'
>  ==> Modified (by you or by a script) since installation.
>  ==> Package distributor has shipped an updated version.
>    What would you like to do about it ?  Your options are:
>     Y or I  : install the package maintainer's version
>     N or O  : keep your currently-installed version
>       D     : show the differences between the versions
>       Z     : start a shell to examine the situation
>  The default action is to keep your current version.
> *** issue.net (Y/I/N/O/D/Z) [default=N] ? 
> 

The BBB is busy chugging away Preparing/Unpacking, so not sure if I'm going to get the 'connman' problem yet with the 16.04.1 build, but wanted to know if the prompts were normal and if I should use the default [N]?

Also, I'm finding that I cannot connect via USB with these builds, must use "ssh ubuntu@arm". Is this to be expected?

goeland86, 2017-04-06:
So, the /etc/issue* file is basically because I updated the message with custom text for Umikaze. It's safe to leave those with N to keep the default text.

As far as USB connection, my experience has been that it needs a reboot to work... I've done my images while connected through USB to not lose connectivity when removing connman (USB interface isn't managed by connman, it's a static IP defined in /etc/networking/interfaces iirc). But for some reason it doesn't always work right away. If you need to, you should be able to do an ifconfig to find the name of the USB interface (something like usb0 maybe?) and run ifup usb0. That should get you going if the reboot solution doesn't work.

Good luck!

bliedblad, 2017-04-06:
GRRR... After a day of pulling my hair out, turns out I had a flaky USB cable. My PC (OSX) would recognize the BBB, mount it as a drive, get an IP from BBB (maybe 10% of the time), then disconnect and restart the cycle.  

In addition to using Kamikaze/Umikaze on my Replicape, I'm trying to learn from you experience to make an image for an embedded machine control system (unrelated to 3d printing).

Question - other than the size difference (2gb vs 4gb), what kind of benefits might I see from running the Ubuntu LTS image kernel instead of the Debian IOT?  Which would you use to get the best performance on the BBB?

Thanks for the help Jon!

goeland86, 2017-04-07:
It's a matter of personal preference I think. I tend to see Ubuntu as having more up-to-date packages than Debian. The release cycle for Ubuntu is a fixed timeline instead of Debian's "it'll be out when it's ready" approach. I originally switched to Ubuntu because of the more up-to-date kernel versions we could select, though on 2.1.0 we're still using a 4.1 kernel, we're aiming for a 4.4 on 2.1.1. I might test a 4.9 release just to see if any major changes are required but that's mainly it.

goeland86, 2017-07-14:
Do you still have questions related to this or can I close this ticket?

BTW be ready for a new beta image this weekend. Fixed the Redeem SPI issue for both BBB and BBB-W, still using a 4.4 kernel.

closed 2017-08-06 by goeland86