intelligent-agent/Refactor #39: Installing Umikaze2

Zwaubel, 2017-05-19:
Hey @goeland86,
I'm having trouble installing the current Umikaze on my BBB (non wireless).
I'm following your guide in the wiki (changed the git link to the renamed repo) and it all works perfectly until the final reboot. When doing this last step, my beaglebone gets stuck on boot with all LEDs lit permanently. Waited for a few minuts but nothing is happening. 
I've been building the image on the last ubuntu flasher image: [https://rcn-ee.com/rootfs/2017-04-07/flasher/BBB-eMMC-flasher-ubuntu-16.04.2-console-armhf-2017-04-07-2gb.img.xz](url)

Is there something missing in the guide?

goeland86, 2017-05-19:
No. I've run into similar problems with the BBBW myself.

I'm trying to work on fixing this, and I think it's related to the kernel update script... Try looking on your SD card's boot/uEnv.txt if the kernel version it's trying to boot is the correct one (use the 4.4.x kernel version from the vmlinuz-4.4.x in boot exactly). This should let you fix that and keep going.

Zwaubel, 2017-05-21:
Thank you for the fast response. I'm now running into another problem: 
It seems like the system partition, created by etcher.io to install the ubuntu image to, is to small for Umikaze. As the title of the image already tells, it creates a 2GB partition. 
I guess the current Umikaze distribution is bigger than that. Is there an easy way to change the partition size on the SD card?

Zwaubel, 2017-05-21:
Okay, I should have read the elinux.org web page more attentively. The script in this section seems to do the magic: http://elinux.org/BeagleBoardUbuntu#All_BeagleBone_Varients

goeland86, 2017-05-21:
The way I do it is to use gparted after I flashed the micro-SD... Should probably add that in the wiki.

Zwaubel, 2017-05-21:
I was thinking of doing that, too. But i currently only have a MS Surface running Windows to work with...
Experiences so far:
- The installation of networkmanager terminates the internet connection and causes the prep_ubuntu script to fail in the following.
- Sometimes I'm getting weird entries for the nameserver in resolv.conf. Mostly something like 127.0.0.1...

goeland86, 2017-05-21:
About the network-manager install - it's a known problem and I run prep_ubuntu while connected through USB

Zwaubel, 2017-05-21:
Okay, I thought, I just let you know.

You were right about that kernel entry in boot/uEnv.txt. It was referencing the kernel 4.4.59-bone17... changed that to the updated one and now it is working. Thanks for that!

closed 2017-08-06 by goeland86