intelligent-agent/Refactor #41: Slic3r-XS-0.01 installation aborted with error

Zwaubel, 2017-06-01:
The latest slic3r does not install correctly. In one of the last steps the following error pops up:

> Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
arm-linux-gnueabihf-gcc -I/usr/lib/arm-linux-gnueabihf/perl/5.22/CORE -fPIC -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -Isrc/eigen -I/usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -std=c++11 -DSLIC3R_GUI -DSLIC3R_PRUS -DUNICODE -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -DBOOST_LOG_DYN_LINK -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/admesh/shared.o src/admesh/shared.c
arm-linux-gnueabihf-gcc -I/usr/lib/arm-linux-gnueabihf/perl/5.22/CORE -fPIC -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -Isrc/eigen -I/usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -std=c++11 -DSLIC3R_GUI -DSLIC3R_PRUS -DUNICODE -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -DBOOST_LOG_DYN_LINK -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/utils.o src/libslic3r/utils.cpp
arm-linux-gnueabihf-gcc: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
error building src/libslic3r/utils.o from 'src/libslic3r/utils.cpp' at /usr/share/perl/5.22/ExtUtils/CBuilder/Base.pm line 173.
! Installing ./xs failed. See /root/.cpanm/work/1496340309.26080/build.log for details. Retry with --force to force install it.
FAIL
1 distribution installed
The XS/C++ code failed to compile, aborting
 

Unfortunately force install does not change the anything and the log file does not contain any more details about the error.

Here is the content of build.log:

> cpanm (App::cpanminus) 1.7043 on perl 5.022001 built for arm-linux-gnueabihf-thread-multi-64int
Work directory is /root/.cpanm/work/1496340309.26080
You have make /usr/bin/make
You have /usr/bin/wget
You have /bin/tar: tar (GNU tar) 1.28
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
--> Working on ./xs
Entering /usr/src/Slic3r/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.06)
Checking if you have Module::Build 0.38 ... Yes (0.4216)
Configuring /usr/src/Slic3r/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Typemaps::Default 1.05 ... No
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... No
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have Test::More 0 ... Yes (1.302085)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have Module::Build 0.3601 ... Yes (0.4216)
==> Found dependencies: ExtUtils::Typemaps::Default, ExtUtils::Typemaps::Basic
Searching ExtUtils::Typemaps::Default (1.05) on cpanmetadb ...
--> Working on ExtUtils::Typemaps::Default
Fetching http://www.cpan.org/authors/id/S/SM/SMUELLER/ExtUtils-Typemaps-Default-1.05.tar.gz
-> OK
Unpacking ExtUtils-Typemaps-Default-1.05.tar.gz
Entering ExtUtils-Typemaps-Default-1.05
Checking configure dependencies from META.json
Checking if you have Module::Build 0.40 ... Yes (0.4216)
Checking if you have ExtUtils::Install 1.46 ... Yes (2.06)
Configuring ExtUtils-Typemaps-Default-1.05
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have Test::More 0 ... Yes (1.302085)
Checking if you have Module::Build 0 ... Yes (0.4216)
Checking if you have ExtUtils::Typemaps 3.18_03 ... Yes (3.30)
Building and testing ExtUtils-Typemaps-Default-1.05
-> OK
Successfully installed ExtUtils-Typemaps-Default-1.05
Searching ExtUtils::Typemaps::Basic (1.05) on cpanmetadb ...
Already tried ExtUtils-Typemaps-Default-1.05. Skipping.
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See /root/.cpanm/work/1496340309.26080/build.log for details. Retry with --force to force install it.
1 distribution installed

goeland86, 2017-06-01:
Hi,

So, known problem when building Slic3r is the amount of memory it needs. Do you have a spare partition on an SD or USB drive that you declare as swap? (i.e. swapon /dev/mmcblk0p2 if you have a 2nd partition on your SD card). The compilation of Slic3r takes more than what Ubuntu can provide in 512MB of ram...

Be aware that whatever partition you use as swap will be written over with seemingly random binary garbage and will need to be reformatted to be useful other than as swap later.

Jon

Zwaubel, 2017-06-05:
Thank you for the fast response. Haven't tried it yet because I ran into another problem: redeem won't start up. Seems like there is a dependency missing and systemctl restart redeem does not work either. It gets stuck on command execution...

here is the status output:

> root@kamikaze:/usr/src/redeem# systemctl status redeem
● redeem.service - The Replicape Daemon
   Loaded: loaded (/lib/systemd/system/redeem.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

>Jun 05 11:14:08 kamikaze systemd[1]: Dependency failed for The Replicape Daemon.
Jun 05 11:14:08 kamikaze systemd[1]: redeem.service: Job redeem.service/start failed with result 'dependency'.

Also: There is no log file being created, that you would be able to take a look at from within octoprint...

goeland86, 2017-06-13:
Hi, can you confirm if running out of memory was indeed the issue or not?

Zwaubel, 2017-06-14:
The installation takes ages, which is an improvement. Without additional swap the installation was terminating prematurely at the point when it started to actually compile slic3r. 
Now it did go far beyond that point but somehow still terminated before it actually finished. (Maybe due to some weird timeout thing because I was not always connected with my main computer via ssh)
The log file does not show any unusal behaviour, so I think it probably would work if I would be connected continously while it is installing.
I'll try again soon (today or tomorrow) and will let you know about my success...

Zwaubel, 2017-06-15:
I'm still struggling with the installation... The compilation of slic3r really takes a long time and at a certain point the beaglebon just crashes. There might be something wrong with my sd card and/or the swap partition that I created on it.. I'll try to work that out...

Zwaubel, 2017-07-13:
Success! Finally it's installed. The key this time was, to set up a partition of no more than 4 GB of swap space.
Before, I just used the whole SD card of 32GB, that probably messed up the address space or something..
Oh, by the way, I had to manually install the octoprint slic3r plugin und configure it. I don't know if this only due to the fact that umikaze did not install correctly on the first attempt, but it would be cool to have that done in the bash script straight away (same as with the cura installation).

Again, thank you for your valuable advice!

goeland86, 2017-07-13:
Cool! Good to know. My SD was 8GB, so I guess that didn't mess up too much of the address space when I used it (plus it was a 4GB partition on it, instead of the whole card).

I'll look into the slic3r-octoprint plugin when I have time to devote to Umikaze2 dev again - I've been coasting for too long...

closed 2017-08-06 by goeland86