intelligent-agent/Refactor #42: The redeem service won't start up after completing the make-kamikaze script

Zwaubel, 2017-06-07:
After executing the make kamikaze script redeem won't start up. Seems like there is a dependency missing and systemctl restart redeem does not work either. It gets stuck on command execution...

here is the status output:

root@kamikaze:/usr/src/redeem# systemctl status redeem
● redeem.service - The Replicape Daemon
Loaded: loaded (/lib/systemd/system/redeem.service; enabled; vendor preset: enabled)
Active: inactive (dead)
Jun 05 11:14:08 kamikaze systemd[1]: Dependency failed for The Replicape Daemon.
Jun 05 11:14:08 kamikaze systemd[1]: redeem.service: Job redeem.service/start failed with result 'dependency'.

Zwaubel, 2017-06-07:
I've been digging may way through the script now and it seems that this error is related to redeems install method (install_redeem()). I will inspect the method tomorrow when I got time.

goeland86, 2017-06-08:
Interesting. Please let me know what you find, because in our latest attempts in the Replicape slack, several others who tried the script didn't report a failure of installation. The dependency failure was instead due to one of the SPI devices missing or being erroneous.

We're still working on making sure we get the right device loaded as part of the redeem.service dependency in systemd, but check that aspect of it?

Zwaubel, 2017-06-09:
And you are right, it seems to be depending on the spi rules, more precisely with this line of the script:
cp scripts/spidev.rules /etc/udev/rules.d/    (line 197 of the make kamikaze script)

When using the old kamikaze 2.1.0b image and manually updating redeem via the git, I've never copied the new rules . So with the old rules at /etc/udev/rules.d/ redeem does run as it should but when I manually copy the new rules to this location and reboot, redeem will fail to load telling me about the dependency error...

I'm currently running a blank beagle bone with only the replicape attached.
Have you been changing your SPI connection rules so that redeem won't start when a certain device is not available or does not respond (correctly)?

PS: I'll now check which rule exactly is causing problems.

goeland86, 2017-06-10:
So the SPI rule was changed to try and accomodate for the different SPI number shown for the replicape board on the wireless version of the beaglebone black, which obviously is causing issues on a non-wireless board. 

There are several tests going on until we commit a fix. I'm also awaiting a development replicape board (one to just test if redeem will start, not the one hooked up to my printer). If you care to, you're welcome to join in the discussion on the replicape slack's #development channel (invite link on the thing-printer.com website)

Once we have a comprehensive solution to fix the issues on both the ethernet and wireless 'bones the script will be updated. In the meantime, I would suggest reverting the SPI device number by hand in your .service file.

Zwaubel, 2017-06-10:
Okay, well.. that is annoying. Changing the numbers manually did work perfectly though.
I'll take a look at the slack channels, thanks!

goeland86, 2017-06-13:
I'm opening an issue specific to the SPI numbering issue and closing this one, unless you have objections?

Zwaubel, 2017-06-14:
Nothing to add. Sounds good to me.

goeland86, 2017-07-14:
FYI I fixed the issue - if you look at the change in the scripts/redeem.service file you'll see how easy it really was... Booted on both BBB and BBBW with no other modification.

closed 2017-07-14 by goeland86