intelligent-agent/Refactor #48: Umikaze 2.1.1 rc3

etrombly, 2017-07-26:
On a fresh install of Umikaze I get the following error:
```
07-26 14:05 root         ERROR    Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Redeem-1.3.1-py2.7-linux-armv7l.egg/redeem/GCodeProcessor.py", line 124, in execute
    self.gcodes[val].execute(gcode)
  File "/usr/local/lib/python2.7/dist-packages/Redeem-1.3.1-py2.7-linux-armv7l.egg/redeem/gcodes/M82.py", line 26, in execute
    if axis not in self.printer.axes_absolute:
AttributeError: Printer instance has no attribute 'axes_absolute'

##################:  XYZEH
python: redeem/path_planner/Path.cpp:52: void calculateLinearMove(int, long long int, long long int, double, std::vector<Step>&): Assertion `stepTime > 0 && stepTime < time' failed.
```

I added back in:
self.axes_absolute = ["X", "Y", "Z", "E", "H", "A", "B", "C"]

to line 99 of Printer.py (it was removed at some point). And it gets rid of the python error, but I'm still failing the assertion in Path.cpp.

Other than that it seems like the octoprint and redeem updates are working now.

goeland86, 2017-07-26:
Thanks for the report Eric - this is Redeem related, but I've passed on the
error to the Redeem guys on Slack - as soon as they've resolved it I'll
indicate you need to pull a fresher copy of redeem to fix the problem.

Cheers,
Jon

etrombly, 2017-07-26:
Switched the redeem branch from develop to staging for now and it's working good. Will switch back once this is fixed. Thanks

goeland86, 2017-08-02:
It should've been fixed in develop now - if you care to switch back. I'm gonna mark this as closed.

closed 2017-08-02 by goeland86