intelligent-agent/Refactor #49: Evaluate the use Pypy to run Octoprint

goeland86, 2017-08-04:
As per discussed in the development channel on the Replicape Slack on 4th Aug. 2017, using Pypy to run Octoprint could yield better CPU performance overall for the printer.

Per what Wiley was saying, mrab-regex needed to be recompiled, with the patch being:

```
diff -r 8d0218a38bb9 regex_2/regex/_regex.c 
--- a/regex_2/regex/_regex.c    Sat Apr 29 23:36:01 2017 +0100 
+++ b/regex_2/regex/_regex.c    Fri Aug 04 08:05:50 2017 +0000 
@@ -52,7 +52,7 @@ 

/* Issue 29943: PySlice_GetIndicesEx change broke ABI in 3.5 and 3.6 branches
 */ 
-#if defined(PySlice_GetIndicesEx) 
+#if !defined(PYPY_VERSION) && defined(PySlice_GetIndicesEx) 
#undef PySlice_GetIndicesEx 
#endif 
 
diff -r 8d0218a38bb9 regex_3/regex/_regex.c 
--- a/regex_3/regex/_regex.c    Sat Apr 29 23:36:01 2017 +0100 
+++ b/regex_3/regex/_regex.c    Fri Aug 04 08:05:50 2017 +0000 
@@ -53,7 +53,7 @@ 
#if PY_VERSION_HEX < 0x03070000 
/* Issue 29943: PySlice_GetIndicesEx change broke ABI in 3.5 and 3.6 branches  
 */ 
-#if defined(PySlice_GetIndicesEx) 
+#if !defined(PYPY_VERSION) && defined(PySlice_GetIndicesEx) 
#undef PySlice_GetIndicesEx 
#endif 
#endif
```

There is an additional self-explanatory error in Octoprint to also correct in the install directory.

danielfmo, 2019-05-02:
Did anyone tried PyPy on octoprint more recently? I would like to know if it's possible and if it actually improves performance.

goeland86, 2019-05-02:
@danielfmo I have not investigated in a long time, no. I need to revisit this, but I don't have enough hours in the day at the moment.

ThatWileyGuy, 2019-05-03:
I evaluated this a few months ago. OctoPrint now runs on PyPy without any changes/patches, but the startup time is significantly worse than CPython.

goeland86, 2020-02-17:
I've tried again today with OctoPrint 1.3.12. Octoprint boot time is normal, and all the UI functions normally. However when trying to print, it generates a blocking error that forces a disconnect.

eliasbakken, 2021-02-12:
I think we should leave this for now. Boot time seems to be the most important factor to consider, so if boot times are slower, it is not really an improvement.

closed 2021-02-12 by eliasbakken