intelligent-agent/Refactor #56: Improvement Request: Randomize root password on install

TechMav75, 2017-09-19:
Recommend that the root password be scrambled at install, to close the security hole.
Documentation has been updated to recommend logging in as the ubuntu user instead of root.

goeland86, 2017-09-20:
I'm thinking it may be worth actually revisiting this where on first login it'll require a new password be set. Would that be sufficient? I like the idea of people having root on their machines, otherwise it'll be a who's who of "did you run with sudo in front of the command?"

TechMav75, 2017-09-20:
Forcing a password change at first login would be good. As for the root password, if someone wants the root, they can sudo and change it. I'm looking at it from the view of the non-savvy who would leave it as the default, which would be a security hole.

goeland86, 2017-09-20:
Well, I'd really like to think that if we force a root password change they're not going to put in the exact same password they were supposed to avoid (and I can force that I think, with some research and tuning). I don't have a strong preference either way really, just think it warrants further discussion 'cuz all the docs and build scripts assume to be run from the root user.

closed 2017-11-12 by goeland86