intelligent-agent/Refactor #58: USB driver problems after some time

goeland86, 2017-09-23:
Investigate and fix problems with the USB driver losing devices after some time.

Devices lost by me:
MS HD3000 webcam
USB wifi dongle (multiple brands)

Devices reported to lose connection by other users:
fiveofds : waveshare touch capability (good picture still, through powered USB hub)

Further logs needed:
dmesg before and after device connection lost
list of modules loaded (lsmod)

fiveofds, 2017-09-23:
[after_screen.log](https://github.com/intelligent-agent/Umikaze/files/1326811/after_screen.log)
[before_screen.log](https://github.com/intelligent-agent/Umikaze/files/1326812/before_screen.log)
[mods_with_screen.log](https://github.com/intelligent-agent/Umikaze/files/1326813/mods_with_screen.log)
[mods_without_screen.log](https://github.com/intelligent-agent/Umikaze/files/1326814/mods_without_screen.log)

fiveofds, 2017-09-24:
ubuntu@kamikaze:~$ lsmod
Module                  Size  Used by
omaplfb                12142  1
spidev                  7160  2
ti_am335x_adc           5340  0
kfifo_buf               3220  1 ti_am335x_adc
industrialio           49177  2 ti_am335x_adc,kfifo_buf
pvrsrvkm              155165  17 omaplfb
w1_gpio                 3316  0
omap_sham              21447  0
uio_pruss               4928  2
omap_aes               13297  0
omap_rng                4572  0
rng_core                7254  1 omap_rng
snd_usb_audio         133925  0
snd_hwdep               5453  1 snd_usb_audio
uvcvideo               64917  1
pwm_tiehrpwm            4706  0
snd_usbmidi_lib        18971  1 snd_usb_audio
videobuf2_vmalloc       4848  1 uvcvideo
videobuf2_memops        2402  1 videobuf2_vmalloc
snd_rawmidi            19758  1 snd_usbmidi_lib
videobuf2_v4l2         19489  1 uvcvideo
snd_seq_device          4674  1 snd_rawmidi
videobuf2_core         22018  2 uvcvideo,videobuf2_v4l2
snd_soc_davinci_mcasp    15894  0
snd_soc_edma            1290  1 snd_soc_davinci_mcasp
snd_soc_omap            3058  1 snd_soc_davinci_mcasp
snd_soc_core          154903  3 snd_soc_davinci_mcasp,snd_soc_edma,snd_soc_omap
snd_pcm_dmaengine       5209  2 snd_soc_core,snd_soc_omap
snd_pcm                79254  5 snd_usb_audio,snd_soc_davinci_mcasp,snd_soc_core,snd_soc_omap,snd_pcm_dmaengine
snd_timer              17941  1 snd_pcm
snd                    57864  8 snd_usb_audio,snd_soc_core,snd_hwdep,snd_timer,snd_pcm,snd_rawmidi,snd_usbmidi_lib,snd_seq_device
soundcore               7029  1 snd
spi_omap2_mcspi        11204  0
evdev                   9971  16
ti_am335x_tsc           5161  0
usb_f_acm               7031  2
u_serial               10116  4 usb_f_acm
usb_f_ecm               9400  2
usb_f_rndis            22035  2
u_ether                10949  2 usb_f_ecm,usb_f_rndis
ti_am335x_tscadc        5572  2 ti_am335x_adc,ti_am335x_tsc
libcomposite           42673  14 usb_f_acm,usb_f_ecm,usb_f_rndis
leds_gpio               3420  0
uio_pdrv_genirq         3437  0
uio                     8678  6 uio_pruss,uio_pdrv_genirq
xt_REDIRECT             1885  1
nf_nat_redirect         1534  1 xt_REDIRECT
xt_tcpudp               3155  1
iptable_nat             1997  1
nf_conntrack_ipv4      14365  1
nf_defrag_ipv4          1677  1 nf_conntrack_ipv4
nf_nat_ipv4             5328  1 iptable_nat
nf_nat                 15004  2 nf_nat_redirect,nf_nat_ipv4
nf_conntrack           91228  3 nf_nat,nf_nat_ipv4,nf_conntrack_ipv4
iptable_filter          1786  0
ip_tables              11493  2 iptable_filter,iptable_nat
x_tables               17359  4 ip_tables,xt_tcpudp,iptable_filter,xt_REDIRECT
tilcdc                 26607  1 omaplfb
tda998x                10786  0
ubuntu@kamikaze:~$

fiveofds, 2017-09-24:
[dmesg-after.log](https://github.com/intelligent-agent/Umikaze/files/1327401/dmesg-after.log)

fiveofds, 2017-09-24:
[lsmod.pdf](https://github.com/intelligent-agent/Umikaze/files/1327411/lsmod.pdf)

fiveofds, 2017-09-24:
dmesg before device connection lost working on catching with connection again

fiveofds, 2017-09-24:
[lsmod-beore loss.pdf](https://github.com/intelligent-agent/Umikaze/files/1327415/lsmod-beore.loss.pdf)

eliasbakken, 2021-08-21:
Is this still a problem? I think we can close this issue now if there is no more activity on it. If it comes up again later, please open a new issue!

closed 2021-08-21 by eliasbakken