intelligent-agent/Refactor #66: Boot speed

pistolero, 2018-02-23:
My BBB with Umikaze 2.1.2 takes 1min 51.916s to boot according to systemd-analyze.

Do you know how to reduce this to some meaningful numbers (less than a minute)?

goeland86, 2018-02-23:
Hi @pistolero unfortunately I don't have a sure-fire solution to speed it up that much. What I can suggest though is to look at what we have enabled and disable features you don't need. For example the CIFS server shares can probably shave a good 10-15 seconds on its own. OctoPrint is usually the slowest-starter, and there's really not much I can do to speed it up, other than to suggest having as few plugins installed as possible.

pistolero, 2018-02-23:
@goeland86 that's what I did. Now it is 1min 40s and I'm still analyzing char built by `systemd-analyze plot` in order to understand how to reduce it further.

So, now basically everybody who uses Replicape just sits for 2 minutes till printer boots?
(I'm also lucky to have B3 board so it is almost 2 minutes of stepper noise)

Octoprint is slowest, but I think it takes more than 2 minutes to start, since when I access it, it says "Still loading" and I must wait another few minutes before I can access UI.

goeland86, 2018-02-23:
@pistolero unfortunately, yes. The BBB is not a super-fast system. It's true I haven't benchmarked with older versions of Umikaze or Kamikaze to compare boot times, and maybe we can do some re-ordering to speed it up, but I haven't yet figured out in detail what it is. I'm planning on doing some big spring cleaning after 2.1.2 is out.

Also, I have a B3 as well, so I understand the fun of the grumbling printer. There's normally a fix to the kernel that can set the registers properly before Redeem starts to quiet it down, but I haven't managed to replicate that part of the work that @eliasbakken did from Kamikaze 2.0.8

pistolero, 2018-02-23:
@goeland86 yes, I've heard about the uboot fix but never saw it in action. 

This is sad, I really like Replicape hardware, it is a beautiful board but those issues are killing positive user experience.

I'm thinking about switching to RPi3 with klipper firmware and my old RAMS board to drive steppers. People are saying that they could boot RPi3 to login prompt in 10 seconds.

But while I'm still here I think I'm gonna try few more things.

By examining [vmstat output](https://github.com/intelligent-agent/Umikaze/files/1751311/vmstat.txt)  while board is booting I see that it spends a lot in iowait. I think by booting it from high-class MicroSD cards I can get significant improvement in a boot speed. This could help with Octoprint startup time as well, as far as I understand internal eMMC is quite slow.

goeland86, 2018-02-23:
@pistolero well, Klipper on RPi3 with a RAMPS board to drive steppers means you have two points of failure instead of one - RPi and RAMPS. But I understand your frustration. You're welcome to try with a fast SD to see if that improves what you need, but I will point out that you can also boot a BBB to login prompt in 15-20 seconds, you just still have the rest of the services starting up in the background. RPi3 is fast, yes, but it's not that fast. We run OctoPrint on RPi2s at the local fablab, and the Pis boot super quick, yes, but OctoPrint still takes a good 2-3 minutes to finish its internal startup procedure.

pistolero, 2018-02-23:
From vmstat I see that boot is mostly CPU/disk bound.

The advantage of the RPi3 is that is has more cores and therefore can do more work in parallel, for instance start more services in parallel.

With BBB I have tried installing X11 to show Octoprint UI on my 7" touchscreen. It worked, but was totally unresponsive up to the moment where Redeem would stop printing from time to time. Maybe I messed something with X11 settings, but I got impression that BBB CPU just not powerful enough to run Chromium in kiosk mode and Redeem at the same time.

With Octoprint I think it does a lot of unnecessary work during startup. It seems that it could be changed to start much faster.

goeland86, 2018-02-23:
@pistolero so yes, CPU and disk bound is correct. The RPi3 has a quad-core 1GHz arm, where we have a single core 1GHz chip. The other difference is the amount of ram, thus swapping, that may need to happen (ideally very little)

Regarding X11/Chrome to show OctoPrint on the BBB itself, that is expected and really not encouraged at all. Redeem, depending on the type of printer, can need up to 40% of CPU cycles. OctoPrint just serving gcode and webpages is known to need up to 30-35% of CPU cycles. There's really not much room left to build in the traditional X11/Chrome/OctoPrint GUI. Toggle is a really lightweight GUI that ties into the OctoPrint API on the backend and uses the framebuffer directly to avoid the whole X11 layer.

As for OctoPrint efficiencies, we know it's not perfect and we are working on ways to at least get consistently faster prints by using the "Print from SD" emulation feature which is the main reason for the 2.1.2 release of Umikaze. 

OctoPrint's repo managed by @foosel is undergoing development for version 1.4.x which should have a modular comm layer, and thus help lighten up the internal code loading along with a significant performance requirement drop while sending GCodes. If you really want to try doing away with OctoPrint, a french dev on the Slack channel has linked me his OctoPrint-free UI for touchscreens on github: https://github.com/FuraX49/atcore-eglfs/releases

We're doing our best to get more feature-full and efficient software out to our users, but aside from @eliasbakken we're all volunteers on the project, and several of us are involved in unrelated startups, which consumes a lot of time. I can see why you want better software, we're trying to build it. But at the moment we're going as fast as we can given the other commitments the dev team currently has.

cwhinton, 2018-02-23:
I've found that I'm able to ssh into the BBB almost immediately after it starts responding to pings.  When I tail -f /var/log/syslog, it's not very long before redeem has completed it's startup. OctoPrint does a large amount of what looks like optimistically caching stuff, checking for updates, looking for gcode files to 'process', etc.  

On my last boot, systemd reported 

> Startup finished in 6.581s (kernel) + 1min 30.799s (userspace) = 1min 37.381s.

and 43 seconds later OctoPrint reported

> octoprint.server.preemptive_cache - INFO - ... done in 46.81s 

Redeem was loaded and ready 50 seconds after power up.  

While I don't track  and record all my restarts, I do tend to watch them via ssh and this 'feels' pretty typical to me.

I'd be looking at things in OctoPrint to optimize.  Perhaps it takes longer the more models are present, etc.

pistolero, 2018-02-23:
@cwhinton yes, it is typical. I feel that those 46.81s caching could be saved between restarts, but I don't know any details about Octoprint internals yet.

goeland86, 2018-02-23:
My gut feeling is that it's not only OctoPrint doing this re-caching stuff, but also the underlying Tornado server and whatever other libraries are used.

cwhinton, 2018-02-23:
I did a quick search on the upstream OctoPrint github and didn't see any issues for slow start up. I don't know if it's faster on other platforms or what, but perhaps they could give insight.

pistolero, 2018-02-23:
It seems that adding 
```
devel:
  webassets:
    bundle: false
```
at the end of the `/home/octo/.octoprint/config.yaml` cuts Octoprint loading time in half for me at the expense of shipping all individual jss/css files separately to the browser (quite fast on the LAN and could be mitigated by adding caching headers). It seems that bundling assets on anemic BBB CPU is not a great idea.

pistolero, 2018-02-23:
I'm pretty sure that with webassets it should be possible to build assets once and store them to avoid constant rebuild, that would be even faster, but I'm still trying to understand what is going on inside Octoprint.

pistolero, 2018-02-23:
I also added `preemptive:false`.

As far as I see Preemptive caching just starts a new thread on startup where simulates page request (which does assets building, template rendering and so on). Theoretically it should prepare a cached version of the page to be served to the visitor.

But since we have just one core and most probably trying to access Octoprint UI from the browser immediately after startup the preemptive cache competes with a request from browser for scarce BBB resources. So I think it is OK to turn it off.

So now end of config looks like this:
```
devel:
  webassets:
    bundle: false
    clean_on_startup: false
  caching:
    preemptive: false
```

I'm getting less than a minute from Octoprint restart to full rendering in the browser.

carlsoar, 2018-02-23:
@goeland86 @pistolero I don't understand linux itself well enough to know, but I assume octoprint waits for graphical.target to finish before starting, which is the biggest time-sink.  I can try with a 90MB/s uSD this weekend to see if these numbers change. 

`systemd-analyze critical-chain
The time after the unit is active or started is printed after the "@" character.
The time the unit takes to start is printed after the "+" character.

graphical.target @40.834s
└─sgx-startup.service @35.261s +5.254s
  └─multi-user.target @35.251s
    └─smbd.service @27.080s +8.164s
      └─nmbd.service @12.260s +14.809s
        └─network-online.target @12.178s
          └─NetworkManager-wait-online.service @9.020s +3.133s
            └─NetworkManager.service @5.004s +3.917s
              └─dbus.service @4.203s
                └─basic.target @4.012s
                  └─sockets.target @4.012s
                    └─acpid.socket @4.005s
                      └─sysinit.target @3.964s
                        └─systemd-update-utmp.service @3.696s +154ms
                          └─systemd-tmpfiles-setup.service @2.535s +291ms
                            └─local-fs.target @2.530s
                              └─run-user-1002.mount @19.308s
                                └─local-fs-pre.target @2.529s
                                  └─systemd-tmpfiles-setup-dev.service @1.715s +
                                    └─kmod-static-nodes.service @619ms +255ms
                                      └─systemd-journald.socket @498ms
                                        └─-.slice @437ms`

`systemd-analyze blame
         14.809s nmbd.service
         13.247s generic-board-startup.service
         12.940s samba-ad-dc.service
         12.555s dev-mmcblk0p1.device
          8.164s smbd.service
          5.254s sgx-startup.service
          4.729s apt-daily-upgrade.service
          4.136s systemd-logind.service
          3.917s NetworkManager.service
          3.643s loadcpufreq.service
          3.506s sysstat.service
          3.340s udhcpd.service
          3.241s networking.service
          3.133s NetworkManager-wait-online.service
          2.622s console-setup.service
          1.681s bb-wl18xx-wlan0.service
          1.263s cpufrequtils.service
          1.210s systemd-user-sessions.service
          1.184s ntp.service
           995ms rsyslog.service
           855ms user@1002.service
           840ms systemd-udev-trigger.service
           744ms capemgr.service
           622ms netfilter-persistent.service
           574ms keyboard-setup.service
           566ms pppd-dns.service
           560ms ssh.service
           551ms systemd-journal-flush.service
           459ms avahi-daemon.service
           412ms systemd-udevd.service`

goeland86, 2018-04-08:
@carlsoar afaik there is no graphical target on the console-version of Ubuntu that we run - it's disabled. It may wait for the framebuffer target but that comes up quickly.

@pistolero I will make an RC image of 2.1.2 with the latest config patch that you suggest, if you could then give me feedback on whether that's acceptable I may ship the final image with those settings added in. Obviously release notes will include the why and how to disable it to get the default behavior back.

eliasbakken, 2021-01-20:
On Recore, the boot-time is dramatically reduced. This does not help Replicape users obviously, but the single old core of the BBB will be a limiting factor without sacrificing a lot of felxeibility (vanilla OctoPrint) 
Recore: 
Startup finished in 3.173s (kernel) + 12.570s (userspace) = 15.744s

closed 2021-01-20 by eliasbakken