intelligent-agent/Refactor #69: Umikaze 2.1.2rc4 has a failed dependency on SPI device

goeland86, 2018-03-23:
See Slack's #support channel - screenshot from viper15c.
Redeem log shows :
```
Failed to add dependency on sys-devices-platform-ocp-481a0000.spi-spi_master-spi1.1-spidev-spidev1.1.device,sys-devices-platform-ocp-481a0
```

cwhinton, 2018-04-08:
I upgraded my corexy printer yesterday.  This looks like it was the first boot after upgrade.

Messages in context
```
Apr  8 02:05:42 kamikaze octoprint[399]: 2018-04-08 02:05:42,703 - octoprint.server - INFO - Intermediary server started
Apr  8 02:05:42 kamikaze octoprint[399]: 2018-04-08 02:05:42,714 - octoprint.plugin.core - INFO - Loading plugins from /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/octo/.octoprint/plugins and installed plugin packages...
Apr  8 02:05:42 kamikaze octoprint[399]: 2018-04-08 02:05:42,731 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
Apr  8 02:05:43 kamikaze systemd[1]: [/lib/systemd/system/redeem.service:2] Failed to add dependency on sys-devices-platform-ocp-481a0000.spi-spi_master-spi1-spi1.1-spidev-spidev1.1.device,sys-devices-platform-ocp-481a0000.spi-spi_master-spi1-spi2.1-spidev-spidev2.1.device, ignoring: Invalid argument
Apr  8 02:05:43 kamikaze systemd[1]: Started ACPI event daemon.
Apr  8 02:05:44 kamikaze systemd[1]: Reloading.
Apr  8 02:05:44 kamikaze systemd[1]: [/lib/systemd/system/redeem.service:2] Failed to add dependency on sys-devices-platform-ocp-481a0000.spi-spi_master-spi1-spi1.1-spidev-spidev1.1.device,sys-devices-platform-ocp-481a0000.spi-spi_master-spi1-spi2.1-spidev-spidev2.1.device, ignoring: Invalid argument
Apr  8 02:05:45 kamikaze systemd[1]: Started Daily apt upgrade and clean activities.
```
There are similar messages in the syslog for dates preceding installation on my BBB as well.

goeland86, 2018-04-08:
@cwhinton does your printer still function though? Could it be an issue during initialization after the first boot only? Is it still appearing after a reboot?

cwhinton, 2018-04-08:
Yes the printer is functioning and printing.  It's rebooted at least once after the install but those are the only instances I see in my syslog after installation.

I do get lots of these though (note they're for the same address(?))
```
Apr  8 14:36:08 kamikaze systemd[1]: Timed out waiting for device sys-devices-platform-ocp-481a0000.spi-spi_master-spi1-spi1.1-spidev-spidev1.1.device.
Apr  8 14:36:08 kamikaze systemd[1]: sys-devices-platform-ocp-481a0000.spi-spi_master-spi1-spi1.1-spidev-spidev1.1.device: Job sys-devices-platform-ocp-481a0000.spi-spi_master-spi1-spi1.1-spidev-spidev1.1.device/start failed with result 'timeout'.
```
Again, they also don't seem to impact the printer function.

cwhinton, 2018-04-08:
I shutdown my BBB and restarted it.  I do not find any new `Failed to add dependency` messages after the restart.

mloit, 2018-04-23:
I'm having the same/similar issues running rc5c. I'm running on a BBB Rev 00C [as read from the eeprom]. Redeem fails to start automatically.

Attached is my syslog from bootup [BBB-syslog]. 

I've also included the additional syslog entries [BBB-syslog-redeem] after executing "systemctl start redeem". Redeem does appear to run correctly after the manual start, despite the SPI timout messages in the syslog.

[BBB-Syslog-Redeem.txt](https://github.com/intelligent-agent/Umikaze/files/1938491/BBB-Syslog-Redeem.txt)
[BBB-Syslog.txt](https://github.com/intelligent-agent/Umikaze/files/1938487/BBB-Syslog.txt)

goeland86, 2018-05-07:
Closing as this is fixed in newer builds.

closed 2018-05-07 by goeland86