intelligent-agent/Refactor #72: Make mjpeg_streamer customizeable

mloit, 2018-04-30:
It would be helpful having the systemd service script invoke a shell script, instead of streamer directly, and have the shell script somewhere accessible for OctoPrint to update through the UI, so that people can customize easily based on their webcam.

goeland86, 2018-05-07:
Agreed, but at this point I've got too little time to take a crack at it.

eliasbakken, 2021-09-03:
mjpeg.service fails on Refactor Recore, so we should find a solution to this.

goeland86, 2021-09-03:
@eliasbakken it only fails if no webcam is plugged in. Just so we're clear. We could modify the systemd script to depend on `/dev/video0`?

eliasbakken, 2021-09-04:
The OP is requesting a way to customize certain things, I imagine that is resolution etc. There are ways to fix this in OctoPrint using a combinatition of plugins etc. but it is not a very elegant solution. I think a better way would be to make a plugin for OctoPrint. That could be a solution for the future, but I think for now we should be satisfied with changing the settings by editing the service file directly. 

There is a PR waiting to be merged that causes the service to not fail if there is no webcam present on Recore, I think that will be enough for now.

closed 2021-09-04 by eliasbakken