intelligent-agent/Refactor #73: OctoPrint does not follow hostname changes for webcam settings

mloit, 2018-05-06:
When the `hostname` is changed, The webcam URL settings in OctoPrint remain at `kamikaze.local` Is there any way this can be modified to automatically follow?

Also for the `snapshot` the URL seems to fail, but if I change it to `http://localhost:8080/?action=snapshot` it works fine. As a bonus this will be immune to any `hostname` changes.

goeland86, 2018-05-07:
umm, will need to check on the snapshot URL question. It was supposed to point at `localhost:8080` as a default already.

As to making a change from kamikaze.local to whatever URL when you change hostname, sadly I can't really change that easily.

mloit, 2018-05-07:
Yeah unless OctoPrint offers a keyword that it internally tracks the hostname, there is not much we can do there. But we can have it default to whatever the hostname is set to during install/upgrade. At least then an upgrade won't break anything.

goeland86, 2018-05-07:
Point, I'll try to add a `sed` call to use the hostname value instead of kamikaze.local. But this would be weird, it would imply that it's overwriting the Octoprint config while preserving the hostname...

goeland86, 2018-05-07:
Ok, done in fafee1ab2a473d3b8f976236415179a19a25dc8d , we'll just have to test if it works as intended.

closed 2018-05-07 by goeland86

mloit, 2018-05-08:
interesting solution, in theory that *could* be put in a script executed on each boot before OctoPrint is started, to re-set the hostname to whatever the current setting is. So basically a change in hostname would require a reboot... but that's not unreasonable. I'll perhaps whip something together to see if it works.