intelligent-agent/Refactor #74: syslog misconfigured

mloit, 2018-05-07:
`rsyslogd` continually reports [roughly every 90 seconds] `4:35:20 kamikaze rsyslogd-2007: action 'action 9' suspended, next retry is Mon May  7 04:36:50 2018 [v8.16.0 try http://www.rsyslog.com/e/2007 ]`

This is the result of an output module suspending itself, which can happen whenever there is a problem talking to the output destination (e.g. a remote server may be down). 

So looking into `/etc/rsyslog.conf` and `/etc/rsyslog.d/50-default.conf` I discovered that in `50-default.conf` the log is trying to be piped through `/dev/xconsole` which does not exist in our image. 

```
daemon.*;mail.*;\
	news.err;\
	*.=debug;*.=info;\
	*.=notice;*.=warn	|/dev/xconsole
```
removing the reference removes the warning.
```
daemon.*;mail.*;\
	news.err;\
	*.=debug;*.=info;\
	*.=notice;*.=warn
```

goeland86, 2018-05-07:
Thanks, will include that change for the final release.

eliasbakken, 2021-02-12:
This is not an issue any more. Nothing about xconsole in rsyslog.conf

closed 2021-02-12 by eliasbakken