intelligent-agent/Refactor #82: mount options outdated

goeland86, 2018-12-04:
The mount calls to bind `/dev`, `/sys` and `/proc` are outdated for 18.04...
https://github.com/intelligent-agent/Umikaze/blob/d8fb12ac363c8bb27a0333921ff6ceb5645eefb2/build-image-in-chroot-end-to-end.sh#L45-L47

I would recommend we replace them with
```
mount --rbind /dev ${MOUNTPOINT}/dev
mount --rbind /sys ${MOUNTPOINT}/sys
mount -t proc none ${MOUNTPOINT}/proc
```

Those are the mount commands given for the Gentoo install in their handbook, which is used to chroot into the system being installed on disk from the running livecd.

Wackerbarth, 2018-12-04:
Actually, those mounts are performed on the HOST and not in the chrooted TARGET.
There are some real issues when the two systems are not the same. To really work properly, there needs to be an emulation layer. If the two systems are the same, then these mounts are probably adequate. However, when they differ, the provided /dev and /sys really need to be complex.

Wackerbarth, 2018-12-05:
Actually, some "emulation" is needed. The fact that the HOST is the same architecture as the TARGET just makes many of the aspects of the emulation a simple pass-thru. However, there are still differences. In particular, you need to account for the fact that in `chroot`, we are not running the kernel that is in the TARGET file system. This is important because any program which depends on kernel properties gets "the wrong answer" when it looks into `/sys` or '/dev` to decide on how it should operate. In particular, these scripts which are updating the versions to `apt-get`, `mkinitramfs`, set up `grub`, etc. base their action on the versions of the kernel that it expects to be active when we `reboot`. Just take a look at all of the stuff that appears in the TARGET's `/boot/`. Somehow, I don't expect that the "linux-odroid-xu3-4.14.83-1-armv7h" kernel will run at all, or that the "linux-am33x-4.19.6-1-armv7h" kernel will give you the control of the Replicape that you expect.

I believe that you are correct in suggesting that a separate `/proc` is appropriate. In our case, the TARGET should never need to know anything about the HOST processes. And the HOST is not actively monitoring the processes within the TARGET.

As to `/dev`, in the `chroot`, the "emulation" can present a selected subset of the available devices. However, in our particular use case, the TARGET program, in particular `mkinitramfs` is enumerating the device tree to decide what contents need to appear in its output. Thus, to get the correct result, we really need to "emulate" the population to match that which will appear when the TARGET is booted.

goeland86, 2018-12-05:
As to the kernel update issue. The apt-get scripts detect they're running in a chroot and only update based on what's in the available filesystem, without depending on the running kernel at all.

This can be seen in the logged output of apt-get when it's installing kernel updates and the like.

Wackerbarth, 2018-12-05:
@goeland86 - Jon, you are wrong. As I attempted to explain, in the `build-...-end-to-end` script, although the programs run in the chroot environment don't operate on either the kernels or (most) devices in the `/dev` or `/sys` space, they do enumerate those spaces  to select which files to include in the image that is produced. Somewhere in the process, inappropriate artifacts remain and they affect the result.

goeland86, 2018-12-05:
@Wackerbarth I maintain that they absolutely do not.

Where the programs normally enumerate the devices for some reason, you get the following error message, forcing the program to use a fallback method to accomplish the original task:
```
update-initramfs: Generating /boot/initrd.img-4.14.71-bone17
cryptsetup: WARNING: failed to detect canonical device of /dev/mmcblk0p1
cryptsetup: WARNING: could not determine root device from /etc/fstab
```

Again - if we've been able to build images on an Odroid before that ran on a Beaglebone, why should that aspect fail now? My entire point was that the mount options for the chroot needed to be updated. Why are you devolving this into questioning the entire chroot setup now?

goeland86, 2018-12-05:
Ok, so, first off: no, we're trimming down the required packages as much as we can and not including "oh someone might want it so let's pull this in" dependencies. Everything that's included was either resulting from an issue on the Umikaze repo, decided based on feedback from a G+ community poll or discussed in #development on slack and the team agreed it was a good combo.

Can we do better in trimming more fat? Yes. We always can. And we gladly accept PRs to fix that.
But in the "perfectly trim hardware controller" you envision us building an image for, we may as well be building an entirely new distribution from scratch. And given the lack of time to invest that is a less than ideal decision in the current dev team's environment. Feel free to do it on your own if you feel it's better that way. But do it in another repository.

The reason we can't fit on a 2GB beaglebone anymore is because of a change in PRU code where instead of writing it in assembly we're now using C. This results in noticeably better code, however the overhead of installing the C-compiler for the PRU is what pushes us past the 2GB limit, and this was when we went from 2.1.0 to 2.1.1. I'm trying to trim down to below the 2GB limit again by removing the onboard slicers from the default image while giving a quick script for those who want to use either Cura or Slic3r to build them onto their image.

As to the confusion between HOST and TARGET systems, on @ThatWileyGuy 's Odroid, it is, in fact Ubuntu 18.04.1 as a HOST.

goeland86, 2018-12-05:
This clearly denotes a misunderstanding of how Redeem works on your part. The PRU C code is recompiled everytime the endstop configuration is changed. It's compiled at every redeem restart as well. This is because until we change the way we communicate with the PRUs, communicating a config file to the firmware is, afaik, impossible.

Again, a work-in-progress that is a long running project that should not delay the push of newer releases in the meantime.

As to the optional components, they are to be installed by the user on the final installed system, and not on the SD image itself. That is not the goal.

goeland86, 2018-12-05:
And again - this is completely off-topic to the initial issue that was raised, and deserves its own issue topic, or a coordination meeting with the rest of the development team.

Wackerbarth, 2018-12-05:
As far as the HOST is concerned, I find it problematic that you expect a particular revision of the OS there. Keeping @ThatWileyGuy's Odroid machine, the Umikaze base, and other HOSTs, like mine, all "in sync" is unreasonable. Our build SHOULD work whether the HOST is ahead or behind the Umikaze that it is building. In theory, it should be a script that might run even on an "x86" emulating the "armv7". But that's asking too much.

Wackerbarth, 2018-12-05:
No, I do understand how Redeem currently works. And recompiling the PRU code during a print "sucks". I thought that part of the reason to move to the `ti` kernel was to get the rproc interface that woud allow us to improve the situation.

goeland86, 2018-12-05:
> As far as the HOST is concerned, I find it problematic that you expect a particular revision of the OS there. Keeping @ThatWileyGuy's Odroid machine, the Umikaze base, and other HOSTs, like mine, [...]

Regardless of which version of an OS you're running, even as far back as the Ubuntu 16.04 those mount options will work. The fact that the current flags used are older and being deprecated by Ubuntu now means they may already be deprecated by other distributions, or at least will be soon. They would only do this if the underlying project is deprecating the flags in the next release.

goeland86, 2018-12-05:
> No, I do understand how Redeem currently works. And recompiling the PRU code during a print "sucks". I thought that part of the reason to move to the `ti` kernel was to get the rproc interface that woud allow us to improve the situation.

If you've followed the work, then you know it's not yet ready for release, and therefore off the table for the current release cycle. And again totally off topic. Create a new issue if you want to lead the effort to reduce the image size.

Wackerbarth, 2018-12-05:
As to "components" in a build, the question is "which are optional?". For example, many users have no video camera on the printer itself. Similarly for the Toggle interface. Removing any such would reduce the base image. Too many options become too "complex" for the user. Where can we draw the lines?  It's a good question for which I have no good proposal.

goeland86, 2018-12-05:
> As to "components" in a build, the question is "which are optional?". For example, many users have no video camera on the printer itself. Similarly for the Toggle interface. Removing any such would [...]

Again this belongs to a different issue/discussion - please create the appropriate thread for this. It isn't here.

eliasbakken, 2021-02-12:
Seems to work fine now.

closed 2021-02-12 by eliasbakken