intelligent-agent/Refactor #93: Consider changing default image to be a non-flasher

Wackerbarth, 2019-01-11:
As someone who often generates and tests installation images, I would prefer to have them NOT enable the automatic flasher script in `/boot/uEnv.txt`. If I want to flash, I can log into the running image and execute the appropriate script.

When we actually distribute a release we MAY want it to be an automatic flasher. But, in the interim, it would save time to have the alternate behavior.

Wackerbarth, 2019-01-11:
Agree except for the way the flag is defined. Eventually, I hope we get to a workflow that makes it all automatic from the `git` repository.  But, in any case, unless the builder specifically overrides it, I think that `flash=True` should apply only to "release" branches and that any generic branch name is, by default, a "non-release". Otherwise you are restricting the choice of branch names for non-release builds rather than restricting them for releases.

goeland86, 2019-01-14:
Fair enough.

eliasbakken, 2020-12-26:
Recore is not a flasher by default. I'm not sure about the other platforms.

closed 2020-12-26 by eliasbakken