intelligent-agent/Refactor #100: connman is still present and running in RC2

ThatWileyGuy, 2019-03-21:
It looks like NetworkManager is definitely running the show, but this still confuses Toggle.

```
root@zack:~# systemctl status connman
● connman.service - Connection service
   Loaded: loaded (/lib/systemd/system/connman.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-03-20 05:31:20 UTC; 1 day 10h ago
 Main PID: 430 (connmand)
    Tasks: 1 (limit: 1030)
   CGroup: /system.slice/connman.service
           └─430 /usr/sbin/connmand -n --nodnsproxy

Mar 21 13:23:59 zack connmand[430]: ntp: adjust (slew): +0.001428 sec
Mar 21 13:41:03 zack connmand[430]: ntp: adjust (slew): +0.001023 sec
Mar 21 13:58:07 zack connmand[430]: ntp: adjust (slew): +0.001342 sec
Mar 21 14:15:11 zack connmand[430]: ntp: adjust (slew): +0.000850 sec
Mar 21 14:32:15 zack connmand[430]: ntp: adjust (slew): +0.001156 sec
Mar 21 14:49:19 zack connmand[430]: ntp: adjust (slew): +0.001359 sec
Mar 21 15:06:23 zack connmand[430]: ntp: adjust (slew): +0.001424 sec
Mar 21 15:23:27 zack connmand[430]: ntp: adjust (slew): +0.000812 sec
Mar 21 15:40:31 zack connmand[430]: ntp: adjust (slew): +0.000945 sec
Mar 21 15:57:35 zack connmand[430]: ntp: adjust (slew): +0.000478 sec
root@zack:~# systemctl status NetworkManager
● NetworkManager.service - Network Manager
   Loaded: loaded (/lib/systemd/system/NetworkManager.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-03-20 05:31:22 UTC; 1 day 10h ago
     Docs: man:NetworkManager(8)
 Main PID: 450 (NetworkManager)
    Tasks: 4 (limit: 1030)
   CGroup: /system.slice/NetworkManager.service
           └─450 /usr/sbin/NetworkManager --no-daemon

Mar 21 15:43:35 zack NetworkManager[450]: <info>  [1553183015.5535] device (wlan0): supplicant interface state: disconnected -> scanning
Mar 21 15:43:44 zack NetworkManager[450]: <info>  [1553183024.2796] device (wlan0): supplicant interface state: scanning -> disconnected
Mar 21 15:48:35 zack NetworkManager[450]: <info>  [1553183315.5396] device (wlan0): supplicant interface state: disconnected -> scanning
Mar 21 15:48:59 zack NetworkManager[450]: <info>  [1553183339.2783] device (wlan0): supplicant interface state: scanning -> disconnected
Mar 21 15:53:35 zack NetworkManager[450]: <info>  [1553183615.5234] device (wlan0): supplicant interface state: disconnected -> scanning
Mar 21 15:54:14 zack NetworkManager[450]: <info>  [1553183654.3060] device (wlan0): supplicant interface state: scanning -> disconnected
Mar 21 15:58:35 zack NetworkManager[450]: <info>  [1553183915.5078] device (wlan0): supplicant interface state: disconnected -> scanning
Mar 21 15:59:29 zack NetworkManager[450]: <info>  [1553183969.2408] device (wlan0): supplicant interface state: scanning -> disconnected
Mar 21 16:03:35 zack NetworkManager[450]: <info>  [1553184215.5445] device (wlan0): supplicant interface state: disconnected -> scanning
Mar 21 16:03:41 zack NetworkManager[450]: <info>  [1553184221.2623] device (wlan0): supplicant interface state: scanning -> disconnected
root@zack:~# cat /etc/dogtag
Umikaze 2.2.1-RC2 Tue Jan 29 19:06:58 UTC 2019
root@zack:~#
```

dpkg also shows that the connman package is present.

goeland86, 2019-03-21:
Definitely an issue there.
I need to check why it's leaving connman in place.

eliasbakken, 2021-02-12:
Connman is not present in latest master for Replicape.

closed 2021-02-12 by eliasbakken