intelligent-agent/Refactor #106: Bundled Umikaze revisions [SOLUTION]

manbehindthemadness, 2019-10-21:
1. NetworkManager:
		The current RealTek rtl8192cu driver bundled in Ubuntu has been noted and confirmed partially broken and unstable. We will have to complile it from scratch using the procedure below.
		Ref: https://askubuntu.com/questions/246236/compile-and-install-rtl8192cu-driver
		
		Installation
		Ensure you have the necessary prerequisites:

			sudo apt-get install linux-headers-generic build-essential dkms
		Clone this repository:

			git clone https://github.com/pvaret/rtl8192cu-fixes.git
		Set it up as a DKMS module:

			sudo dkms add ./rtl8192cu-fixes
		Build and install it (this version number may change so the command might need adjustment)

			sudo dkms install 8192cu/1.11
		Refresh the module list:

			sudo depmod -a
		Ensure the native (and broken) kernel driver is blacklisted:

			sudo cp ./rtl8192cu-fixes/blacklist-native-rtl8192.conf /etc/modprobe.d/
		Reboot the system
2. Auto logging dummy user.
		In the newer Ubuntu kernels the rungetty system has changed slightly. In stead of directly altering the execution statement within getty@tty1.service, we need to revert that file to it's
		original contents here: https://github.com/japgolly/.golly-rc/blob/master/ref/laptop-arch/etc/systemd/system/getty.target.wants/getty%40tty1.service
		Please note that this is only needed in the event that the old auto logging user setup script has already been run on the system.
		The instructions to configure the current rungetty auto logon functionality are located here: https://askubuntu.com/questions/819117/how-can-i-get-autologin-at-startup-working-on-ubuntu-server-16-04-1/819154#819154
		In addition to the above I have altered the setup script to complement these instructions (However I haven't tested them yet) Also note that I left the old commands present but commented for comparison:
		
		install_dummy_logging() {
		apt-get install rungetty
		useradd -m dummy
		usermod -a -G systemd-journal dummy
		echo "setfont Lat15-Terminus12x6.psf.gz" >>/home/dummy/.profile
		echo "clear" >> /home/dummy/.profile
		echo "journalctl -f" >> /home/dummy/.profile
		# text='ExecStart=-/sbin/getty -a dummy 115200 %I'
		# sed -i "/.*ExecStart*./ c $text" /etc/systemd/system/getty.target.wants/getty@tty1.service
		mkdir /etc/systemd/system/getty@tty1.service.d
		touch /etc/systemd/system/getty@tty1.service.d/override.conf
		ed -s /etc/systemd/system/getty@tty1.service.d/override.conf <<EOT
		li
		[Service]
		ExecStart=
		ExecStart=-/sbin/agetty --noissue --autologin myusername %I $TERM
		Type=idle
		EOT
		mv /etc/systemd/system/getty.target.wants/getty@tty1.service /lib/systemd/system/
		systemctl enable getty@tty1.service
	}
3. Force Login Octoprint Plugin:
		The Force Login plugin has a runtime but causing it to flood the system logs with repetitive errors and thus should most likely be disabled or removed.
4. Console session font:
		In Ubuntu 18.x the console font is nothing short of huge, this makes reading the log output a nightmare. 
		To resolve this simple add the line FONT=Lat15-Terminus12x6.psf.gz into /etc/default/console-setup.
5. Color markup support:
		In Ubuntu 18.x colorized terminal markup is disabled by default. to resolve this simply add the line force_color_prompt=yes into /home/dummy/.bashrc
6. Startup performance:
		I have already layed out the details of this here: https://github.com/intelligent-agent/Umikaze/issues/105, however, I will re-state them here for completeness.
		We can prevent a 4-5 minute delay caused by the system searching for a USB mass storage file by adding the line USB_IMAGE_FILE_DISABLED=yes into /etc/default/bb-boot.
		We can prevent a 4-5 minute delay caused by network startup timeout by changing the line TimeoutStartSec=5min to TimeoutStartSec=5s in /etc/systemd/system/network-online.target.wants.
7. Screen Rotation:
		This step is entirely optional, I am including it as in my experience most people are using wide screen format displays in landscape mode.
		To accommodate for this we would alter the /boot/uEnv.txt file changing the default rotation in the cmdline to "4".

goeland86, 2019-10-21:
Thanks for this. Testing a build of Umikaze with those fixes included. If they don't break the image, they'll be included in the 2.2.1 release :)

goeland86, 2019-10-23:
So, the build failed... In part because apparently the dkms by default will build against the currently running kernel. Which is not the one that **will be** running on the image after it's been completed. Need to figure that bit out.

manbehindthemadness, 2019-10-23:
I am sure there is a way to pre-compile it into a package and then just deploy it. I will have a look.

manbehindthemadness, 2019-10-24:
Yes this is the solution right here: https://vincent.bernat.ch/en/blog/2018-packaging-driver-debian-dkms

manbehindthemadness, 2019-10-26:
Why don't we just bind it to a startup script for execution on the initial boot?

manbehindthemadness, 2019-10-26:
Alternatively would we be able to include the package into the image after the build script but before distro-image creation?

goeland86, 2019-10-26:
> 
> 
> Alternatively would we be able to include the package into the image after the build script but before distro-image creation?

No. The build-script generates the distro-image file directly... I'm working out some different kinks out of the build scripts now, will try to use your idea of a bundled startup script after that's ironed out.

goeland86, 2020-12-22:
@manbehindthemadness is this still an issue for you in the latest ReFactor images, or have you not really looked at those recently?

goeland86, 2021-02-28:
I'm going to close this, as there has been no response from the original poster, and I'll assume the problem is no longer applicable.

closed 2021-02-28 by goeland86