intelligent-agent/Refactor #124: Clean up and move playbooks into separate folder

eliasbakken, 2020-12-22:
There are quite a few playbooks floating around in the root folder. These should be removed if they are no longer in use and those remaining should be placed in a separate folder.

goeland86, 2020-12-22:
Actually they will be used - the point was to try and provide a number of playbooks that can be used to modify the existing image as needed, or to run upgrades, for example.

eliasbakken, 2020-12-22:
Is "SYSTEM_redeem_octoprint-LEGACY.yml" still in use also? I think it would be good to have as little "clutter" as possible around to not confuse future users.

goeland86, 2020-12-22:
That one isn't, no. We can move it to a legacy folder indeed.

Though I think Redeem may revive at some point - Klipper has certain a few limitations that we may eventually run into, and @ThatWileyGuy seemed quite attached to Redeem. I see no harm in maintaining a way to use ReFactor with Redeem in the long run.

eliasbakken, 2020-12-22:
Sure! Would be great to revive it if Klipper has limitations or even for fun, but if it is not needed now we can simply delete it. We have it in the git history regardless.

goeland86, 2020-12-22:
True. Wouldn't be hard to re-create from scratch either.

eliasbakken, 2020-12-25:
It seems like it is not straight forward to place playbooks in separate folders. Though I'm not so sure about the pattern of having different playbooks for different scenarios, so we might want to look into that pattern.

eliasbakken, 2021-01-27:
This was meerged with  PR #148

closed 2021-01-27 by eliasbakken