intelligent-agent/Refactor #125: Update make-kamikaze.sh to work for different platforms

eliasbakken, 2020-12-22:
The script is meant to be used to run playbooks locally as far as I understand. If that is correct, it should pull in the right 
variables and specify platform in the same manner as the "build-image-in-chroot-end-to-end.sh" script.

goeland86, 2020-12-22:
So - the build-image-in-chroot-end-to-end.sh script exports variables that are then valid in the environment for the make-kamikaze.sh script... Or I think I may have modified it to read command-line parameters for the script as well, didn't I?

If not, then yes it needs to be fixed, but otherwise it should be ok. We also need to rename it to `make-refactor.sh` as well.

eliasbakken, 2021-01-19:
I'm getting this error: 
ERROR! the playbook: system_klipper_octoprint-DEFAULT.yml could not be found
I think the name is wrong.

eliasbakken, 2021-01-19:
PR #147 solves this.

closed 2021-01-20 by eliasbakken