intelligent-agent/Refactor #126: Recore: Add the latest u-boot to the image

eliasbakken, 2020-12-22:
Recore requires a custom u-boot in order to boot properly. 
Right now, this is manually "dd"ed to the first sectors of the image after it has been flashed. 
This should instead be done as part of the Ansible scripts. 
Here is some sudo code to show what is going on. 
```
card="/dev/mmcblk0"
uboot="u-boot-sunxi-with-spl.bin"
dd if=/dev/zero of={card} bs=1k count=1023 seek=1
dd if={uboot} of={card} bs=1024 seek=8 conv=notrunc
```

goeland86, 2020-12-22:
erm.

I don't think this can be done in ansible. Ansible only runs within the chroot itself.
We would need to run this as part of the `build-image-in-chroot-end-to-end.sh` after the image is created, no?

Or can you run this on the recore while it's running from the `${card}` device mounted to /?

eliasbakken, 2020-12-25:
It's OK to update u-boot on the same device as the root filesystem I think, but I'm not sure how to get the randomly generated lo device from within ansible, so I think it's better to do it in the build script for now.

eliasbakken, 2020-12-25:
PR #128 fixes this.

closed 2020-12-25 by eliasbakken