intelligent-agent/Refactor #140: Toggle octoprint key exchange on first boot.

eliasbakken, 2021-01-18:
Octoprint API key is generated on first time run, but this key is needed by toggle to get events. 
Also, a new user with credentials should be made for toggle, but since the "password" is unique it should be generated on first boot 
and placed in both the octoprint config and in the toggle local file.

goeland86, 2021-01-18:
Maybe we can modify toggle to parse it from the octoprint config file?

eliasbakken, 2021-01-18:
I guess we could have it as a part of Toggle itself, but then we would have to guarantee that the user toggle is running as has access to the octoprint config file and we would have to rely on a static location for the octprint confg file. 

I know that BeagleBone uses a "first run" script to do some maintenance chores as the fist run, and it looks like 
armbian-firstrun.service does the same thing. Regenerating ssh keys and other tweaks.
The last command is to disable itself, which seems like a good idea. 

We could generate a similar script for toggle (or perhaps for Refactor). It can be more specific to the platform. 

A script like that could also check for the presence of and which Manga Screen and enable/disable toggle and set up the proper resolution which on an embedded platform should not need to be run on every boot.

goeland86, 2021-01-18:
Sure, but would that not be part of toggle then? Or toggle installation from ansible?

The octoprint config file location could be specified as part of the toggle config

eliasbakken, 2021-01-19:
PR #144 solves this.
If someone comes up with a better solution, we can always merge that.

eliasbakken, 2021-01-20:
#144 merged

closed 2021-01-20 by eliasbakken