intelligent-agent/Refactor #167: Enable Toggle on Replicape

eliasbakken, 2021-02-21:
Toggle is now disabled on BeagleBone. Here are my thoughts on the matter: 

Ok, so I have to give up for now I think. Kernel 4.14 seems bound to sgx 1.14.3699939. If the kernel is fixed we can still choose either Wayland or EGLnative backends, but I've not gotten any of them to work. Weston/Wayland requires a wrapper for libgbm that implements gbm_bo_get_offset etc. Running on EGLnative needs to map /dev/dri/card0 to card1, and does start the application, but spits out: : Cogl-WARNING  Failed to set crtc mode 720x1280: Invalid argument

eliasbakken, 2021-02-21:
Here is a library that seems to deal with a similar issue for Lima: https://github.com/heiher/libmali-rk3399

eliasbakken, 2021-08-15:
I did manage to get Weston to work on BeagleBone Black using the instructions on elinux.com:
https://elinux.org/BeagleBoneBlack/SGX_%2B_Qt_EGLFS_%2B_Weston
Several packages had to be installed manually. To test:
```/etc/init.d/rc.pvr start```
and
```rm /dev/dri/card0```
```ln -s /dev/dri/card1 /dev/dri/card0```

The resolution must be a multiple of 32. For MS2 small add this to uEnv.txt
```video=HDMI-A-1:736x1280```
And the same resolution must be specified in the [core] section of weston.ini

goeland86, 2021-08-16:
Will be interested to try it out as soon as I can get some spare time.

eliasbakken, 2022-06-09:
Since we have stopped development on Replicape for now, I'm closing this.

closed 2022-06-09 by eliasbakken