intelligent-agent/Refactor #168: Recore doesn't boot after apt upgrade

goeland86, 2021-03-01:
It looks like we need to blacklist grub install.

Multiple users have reported issues of recore not booting after running `apt upgrade` on the boards from 3.0.0-RC4 or later images.

eliasbakken, 2021-05-26:
In the latest master, all packages are up to date. I think we need to know which packages are upgraded in order to get any further with this.

eliasbakken, 2021-08-13:
I've tried an "apt update" on the latest image, and since we are updating all packages as a part of the build process, there were no packages to update. If there was a problem with 3.0.0-RC4, I thin kit must be gone now. I'm marking this done for now. If it reappears, we can take another look at it.

closed 2021-08-13 by eliasbakken