intelligent-agent/Refactor #187: Unknown command: tmcuart_send - error is frequently caused by running an older version of the firmware on the micro-controller (fix

iNDeX999, 2021-08-19:
Hi!

I startet to test the board today but unfortunately after a reboot the users disappeared from octoprint 
```
Password:
Repeat for confirmation:
User mwe does not exist!
(venv) root@recore-0137:/home/debian/OctoPrint# octoprint user list
0 users registered in the system:
```
and I had to delete the user.yam to do a setup again. Unfortunately then other default settings are used and there are no more paths or scripts or api keys.

Further I activated the steppers I need in the pinter.cfg. 
After a new reboot with trying to home one of the axes:

```
Send: G91
Recv: // Unknown command: tmcuart_send
Recv: // This type of error is frequently caused by running an older
Recv: // version of the firmware on the micro-controller (fix by
Recv: // recompiling and flashing the firmware).
Recv: // Once the underlying issue is corrected, use the "RESTART".
Recv: // command to reload the config and restart the host software.
Recv: // Protocol error connecting to printer
Recv: !!! Unknown command: tmcuart_send
Recv: ok

```
But I can't find any information about this in the wiki 

Greetings 
Markus

eliasbakken, 2021-08-20:
I've gone over this on Discord, but for future reference if anyone comes across this: 
The ```tmcuart_send``` command is not present in version V3.0.0 of Refactor. Support for TMC2209 stepper drivers were added with PR #178 / #179. If you are using V3.0.0, which is what many Recore boards are shipped with, configuration of TMC2209 needs to be done in the start-up script. This is cumbersome, so the best solution is probably to update Refactor to a more recent version. I've started on instructions for upgrading, but as of now, they are not finished: https://wiki.iagent.no/wiki/Refactor#Upgrading_Refactor_from_the_command_line

eliasbakken, 2021-09-19:
As for the users not existing when running the command   `octoprint user list`, it is because octoprint runs as user `debian` so you need to be user debian to get the list. 

I think I'll close this issue for now, it seems like a solved problem.

closed 2021-09-19 by eliasbakken