intelligent-agent/Refactor #205: eMMC can sometimes get corrupted.

eliasbakken, 2021-08-29:
The eMMC can get corrupted. This message occurs during boot:
```Failed to load '/boot/dtb/allwinner/sun50i-a64-recore-a5.dtb'
libfdt fdt_check_header(): FDT_ERR_BADMAGIC```
The solution for now is to boot from a USB stick and reflash the eMMC. 
This is cumbersome and time consuming and the user looses his settings. 

Some investigation should be done as to whether this is a hardware issue or not. 
If this is expected (MMC cards do get corrupted from time to time) we must consider ways to 
mitigate the issue with software, either using the "power lost" detection for graceful shutdown or 
mounting one partition as read-only.

goeland86, 2021-09-01:
Does this happen in case of a proper shutdown of the board, or have you only observed it when doing abrupt loss of power?

goeland86, 2021-09-01:
One potential fix for this would be to make `/boot/` a separate partition that only gets mounted read-only, so there's no chance of the files being corrupted.

eliasbakken, 2021-09-04:
This might be relevant. I discovered today that a regular reboot failed. Below is the transcript. Turning the power off and then on again led to a regular boot. It might be related to the "high power" domain being on through the reboot. 
```U-Boot SPL 2021.10-rc1-05752-g6d3636550c-dirty (Aug 13 2021 - 23:48:15 +0200)
DRAM: 1024 MiB
Trying to boot from MMC2
NOTICE:  BL31: v2.5(release):v2.5-294-gabde216dc-dirty
NOTICE:  BL31: Built : 23:42:35, Aug 13 2021
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x2097e68, model: Recore-A5
ERROR:   MHU: Unexpected protocol (MHU status: 0x0)

U-Boot 2021.10-rc1-05752-g6d3636550c-dirty (Aug 13 2021 - 23:48:15 +0200) Allwinner Technology

CPU:   Allwinner A64 (SUN50I)
Model: Recore-A5
DRAM:  1 GiB
MMC:   mmc@1c11000: 1
Loading Environment from FAT... Unable to use mmc 1:1... In:    serial
Out:   serial
Err:   serial
Net:   phy interface7
eth0: ethernet@1c30000
gpio: pin PH0 (gpio 224) value is 1
gpio: pin PH1 (gpio 225) value is 1
gpio: pin PH2 (gpio 226) value is 1
gpio: pin PH3 (gpio 227) value is 1
gpio: pin PG9 (gpio 201) value is 1
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found U-Boot script /boot/boot.scr
3202 bytes read in 2 ms (1.5 MiB/s)
## Executing script at 4fc00000
U-boot loaded from eMMC or secondary SD
Boot script loaded from mmc
204 bytes read in 1 ms (199.2 KiB/s)
28645 bytes read in 4 ms (6.8 MiB/s)
3821 bytes read in 3 ms (1.2 MiB/s)
Applying kernel provided DT fixup script (sun50i-a64-fixup.scr)
## Executing script at 45000000
10126889 bytes read in 216 ms (44.7 MiB/s)
Failed to iterate over directory Image
Failed to load '/boot/Image'
Bad Linux ARM64 Image magic!
SCRIPT FAILED: continuing...
Scanning disk mmc@1c11000.blk...
Found 2 disks
No EFI system partition
BootOrder not defined
EFI boot manager: Cannot load any image
MMC Device 0 not found
no mmc device at slot 0
```
...

eliasbakken, 2021-09-13:
I had an incident of the device failing in a similar way reported by Index, 
``` ** fs_devread read error **
libfdt fdt_check_header(): FDT_ERR_BADMAGIC```
After a boot with a USB drive, the problem seems to have been corrected by itself, which is curious.

eliasbakken, 2021-09-15:
This error came up today which might be relevant for the debugging:
```
Card did not respond to voltage select! : -110
MMC Device 0 not found
no mmc device at slot 0
starting USB...
```
This looks like a hardware issue. It was done on recore-0102, one of the earliest boards of rev A5.

eliasbakken, 2021-09-16:
A possible workaround if the A5 dtb file is corrupted is to use the A4 dtb file. There is no difference in the device tree between those two revisions.

eliasbakken, 2021-09-17:
I've converted to btrfs on a board and tried to abuse it in order to provoke a corruption. So far I have not been able to corrupt the system by switching the power off. I will try to create an image for others to test also.

eliasbakken, 2021-09-20:
I've spent the weekend getting Refactor to create btrfs images. It is now an option that we can play around with more, but it looks like there is the same problem with files being corrupted. And it seems if that happens on a file that is needed to boot, we get into trouble. For instance /boot/armbianEnv.txt tends to get corrupted easily. It seems like it is reproducible by booting a fresh install, getting to the getting started wizard of OctoPrint and then cutting power. Two files seem to get corrupted easily: armbianEnv.txt and config.yaml for OctoPrint. Booting to a USB stick and running `btrfs check  /dev/mmcblk0p1` has at least in one instance shown that there is a problem with config.yaml. Here is a summary: 
```
root@recore:~# btrfs check  /dev/mmcblk0p1
Opening filesystem to check...
Checking filesystem on /dev/mmcblk0p1
UUID: 5f3f98f7-1709-4eae-9d7e-e472b4da3366
[1/7] checking root items
[2/7] checking extents
[3/7] checking free space cache
[4/7] checking fs roots
root 5 inode 151303 errors 200, dir isize wrong
root 5 inode 420589 errors 1, no inode item
        unresolved ref dir 151303 index 19 namelen 11 name config.yaml filetype 1 errors 5, no dir item, no inode ref
root 5 inode 420590 errors 1, no inode item
        unresolved ref dir 151303 index 21 namelen 11 name config.yaml filetype 1 errors 5, no dir item, no inode ref
root 5 inode 420592 errors 1, no inode item
        unresolved ref dir 151303 index 23 namelen 11 name config.yaml filetype 1 errors 5, no dir item, no inode ref
root 5 inode 420602 errors 1, no inode item
        unresolved ref dir 151303 index 25 namelen 11 name config.yaml filetype 1 errors 5, no dir item, no inode ref
root 5 inode 420603 errors 1, no inode item
        unresolved ref dir 151303 index 27 namelen 11 name config.yaml filetype 1 errors 5, no dir item, no inode ref
root 5 inode 420604 errors 1, no inode item
        unresolved ref dir 151303 index 29 namelen 11 name config.yaml filetype 1 errors 5, no dir item, no inode ref
root 5 inode 420605 errors 1, no inode item
        unresolved ref dir 151303 index 31 namelen 11 name config.yaml filetype 1 errors 5, no dir item, no inode ref
root 5 inode 420606 errors 1, no inode item
        unresolved ref dir 151303 index 33 namelen 11 name config.yaml filetype 1 errors 5, no dir item, no inode ref
root 5 inode 420615 errors 1, no inode item
        unresolved ref dir 151303 index 35 namelen 11 name config.yaml filetype 1 errors 5, no dir item, no inode ref
ERROR: errors found in fs roots
found 3691233280 bytes used, error(s) found
total csum bytes: 3614860
total tree bytes: 117735424
total fs tree bytes: 112640000
total extent tree bytes: 950272
btree space waste bytes: 17969808
file data blocks allocated: 1812430602240
 referenced 3220619264
```

goeland86, 2021-09-20:
I think I'll stress again how f2fs is designed for flash/emmc systems, where unplanned power shortages are expected.

The other main issue here with the file being corrupted points to an application or os level issue: the files should be read without being written to.

Though in the case of the config file during setup, it will be written to. I'm not sure why the armbianEnv gets written to, however.

eliasbakken, 2021-09-20:
Ok, so after creating a second, RO partition for Recore, I've gotten the build system to work again. I've done a print, but had to manually change some settings. I'll commit the changes and we can give this a shot. I've not seen the same corruption to /boot/armbianEnv.txt or to /home/debian/.octoprint/config.yaml that I saw previously. This might be due to the journal now being recovered correctly or it might simply be because I've been lucky so far. Regardless, I think that an RO boot partition is the way forward, so I want to explore this direction further. It implies few changes to our current build system since both partitions are ext4, and it seems like it is necessary to have a functioning boot partition in order to do recovery. 

On a related note, it seems there is a startup script that changes the armbianEnv.txt file and adds some USB storage quirks etc. That is probably the reason why that file had a tendency to get corrupted. In combination with commit=600 in the /etc/fstab, that seems like a recipy for disaster.

closed 2021-11-02 by eliasbakken