intelligent-agent/Refactor #211: Jenkins builds fail due to loop device being in use after unmount

goeland86, 2021-09-04:
The latest set of PR builds have failed to complete because of the following error:
```
+ echo 'Defragmenting partition mounted at /tmp/Refactor-sd.VR0D7U.'
Defragmenting partition mounted at /tmp/Refactor-sd.VR0D7U.
+ e4defrag -c /tmp/Refactor-sd.VR0D7U
+ umount -l /tmp/Refactor-sd.VR0D7U
+ echo

+ echo 'Resizing filesystem.'
Resizing filesystem.
+ e2fsck -f /dev/loop0p1 -y
e2fsck 1.45.7 (28-Jan-2021)
/dev/loop0p1 is in use.
e2fsck: Cannot continue, aborting.
```

This seems to be related to how the chroot is unmounted, and this needs to be fixed as the issue can randomly appear on any build system.

eliasbakken, 2021-09-04:
Are you sure it's OK to force whatever program is using the file to be killed? It looks like it could have adverse effects that can be hard to track down.

goeland86, 2021-09-04:
I'm pretty sure at the point where we're trying to unmount the partition, whatever is using it is not something that we want to keep running. I'm thinking the klipper service it something like it that got started by the ansible scripts that didn't stop for some reason.

eliasbakken, 2021-09-04:
In all the cases you have changed, you are doing a defrag and mounting the the partition a couple of lines above it. How could that start any services?

goeland86, 2021-09-04:
The crash was before the defrag started because it was not unmounted properly.

PR #212 (2021-09-04): Adding checks and force kill processes to unmount
Some processes may not shut down properly when the script attempts to umount.

This should fix #211 for good.
closed 2021-09-04 by goeland86