intelligent-agent/Refactor #239: During flashing using the Octoprint web interface, the UUID exchange is not working as it should

eliasbakken, 2021-09-30:
Sometimes the `/sbin/flash-recore` does not finish as it should. 
It leaves the system in a broken state where both the eMMC and the USB drive have the same UUID.

goeland86, 2021-10-01:
Am I reading it right that you're basically extracting the image file directly to the disk from the `flash-recore` script? In that case you can't avoid it. The only way you could would be to mount the .img to a tmp drive and run an `rsync` between the two, like we used to do for Umikaze on BBB...

eliasbakken, 2021-10-02:
What can not be avoided? 
The script works fine some of the times, the UUID is exchanged after the flashing. But sometimes it seems to get cut off. It might be that the progress is reported to the user before the script has finished, or it might be that all changes have not been written to file with the commands in the script. So if the user is quick to reboot after flashing, that causes the error.

goeland86, 2021-10-02:
The uuid from the image being written to disk.

It may be that some of the disk writing isn't properly flushed out of cache, I think it was either `flush` or `sync` that was responsible for that. Maybe add that before modifying the uuid?

eliasbakken, 2021-10-02:
The disk image will be the same yes, that we cannot avoid. But if you look at the script, it is changed after being flashed. 
So yes, i think a `sync` or a `flush` somewhere would help. Would not a `mount` and an `unmount` also save all changes to disk?

goeland86, 2021-10-02:
Possibly, not entirely sure. The unmount could fail if the cache flush isn't triggered properly. Let me see if I can read the doc a bit more.