intelligent-agent/Refactor #240: Consider running klipper as Debian

eliasbakken, 2021-09-30:
Klipper is now running as user root. This might have security implications, and it is interfering with running 
Moonraker as user debian. 
If Klipper is run as root, it cannot call some of the startup-scripts that it needs to.

eliasbakken, 2021-09-30:
It is possible to run klipper as user and run gpioget commands with sudo, so we should consider that.

shuffleznl, 2021-10-07:
Config interactions (Save_variable and Save_config) make Klipper user (root) owner of files that are accessed/shared by processes (Moonraker/Octoklipper) from other users (debian).

eliasbakken, 2021-11-19:
It would be good to remove the `gpioget` commands from the service file. A PR for that has been added to our fork of klipper

eliasbakken, 2021-11-20:
Running Klipper as debian has been merged with PR #255

closed 2021-11-20 by eliasbakken