intelligent-agent/Refactor #308: Add support for printing from USB drive

eliasbakken, 2022-04-28:
It should be possible to add a thumb drive to a printer and mount it and make the files or folder structure available to Mainsail.

goeland86, 2022-04-29:
Maybe make an automount setup that automatically creates the mountpoint folder in `/usr/share/models` with the name of the USB device? We can make this work on Octoprint as well, and it would become platform generic.

eliasbakken, 2022-05-02:
Yes, good idea to have it work for both mainsail and OctoPrint, but I think we might want to choose a different folder for OctoPrint. It's not a location that follows any standard, so perhaps both MainSail and OctoPrint can use /home/debian/gcode_files?

eliasbakken, 2022-05-03:
It's not so easy to get this working with OctoPrint. If the USB drive that was used to install the new revision with is left in the USB port during a reboot, OctoPrint crashes. It seems to be related to the depth of the tree of the filesystem. So even when with just mounting partition 1 which is the boot partition, OctoPrint fails. 
One possible solution is to mount the disk in a different location, check the content of the drive and then choose action based on whether it is a Refactor drive or a g-code files drive. 
Another option is to not add the udev rule until mainsail is installed.

eliasbakken, 2022-05-07:
Ok, so for now the last option was chosen. It has been tested and merged in main.

closed 2022-05-07 by eliasbakken