intelligent-agent/Refactor #316: Add support for I2C on the STM32

eliasbakken, 2022-06-16:
There is peripheral support for I2C on the STM32F031 and the pins have been broken out on the MCU header. 
But the Klipper code does not yet support I2C. This should be added.

brotherdust, 2022-06-16:
To clarify, I2C is supported, but:
- I2C read isn't implemented (https://github.com/intelligent-agent/klipper/blob/839fb8d56618aea31393e588432bad1f724e6777/src/stm32/stm32f0_i2c.c)
- Adding the I2C GPIO functions to the firmware causes a ROM overflow

eliasbakken, 2023-06-28:
In order to get this working without too many changes, I suggest we enable it for the A7 which has 32 KB of flash instead of 16 KB. It should be possible to change the flash size in the config for Klipper, enable i2ccmds.c and add a second .bin file to /opt/firmware that can in turn be flashed during the first boot. The flash-stm32 script can determine if we have an A7 and flash with 32 KB version or A6 and earlier and flash with 16 KB version.

eliasbakken, 2023-10-30:
Merged in master

closed 2023-10-30 by eliasbakken