intelligent-agent/Refactor #318: BUG, after reflash on recore from USB to eMMc with latest mainsail image file

Jasontwd, 2022-06-19:
Hi Guys,
Sorry, new to reporting bugs, hope this makes sense
Using the latest image:
https://github.com/intelligent-agent/Refactor/releases/tag/v3.1.0-RC0

Using this as instructions:
https://wiki.iagent.no/wiki/Refactor -> Upgrading Refactor from the command line. You're already running from USB I think so you can skip to the second command immediately

Problem found:
did wget [the new file]
the old firmware is still on eMMc and still as klipper server script errors
but reads:
v3.1.0-RC0

Comment from @geoland86
yeah, it should be in /etc/refactor.version
this is my current installation:
root@recore:~# cat /etc/refactor.version
v3.0.3-RC5v3.0.3
 2022-02-19root@recore:~#
As you can see, a bit behind on the times
If yours reads v3.1.0-RC0, then we know the image created by @elias has a probable bug in it

Jasontwd, 2022-06-19:
[image]

eliasbakken, 2022-07-19:
Hi, I'm not sure what has happened here, but it looks like something has gone wrong during the flashing. Could you try the instructions explained in the Reflash image? http://feeds.iagent.no/images/reflash/

eliasbakken, 2022-07-19:
Instructions here: https://wiki.iagent.no/wiki/Reflash

closed 2022-07-26 by eliasbakken