intelligent-agent/Refactor #322: mainsail image - path and folder change

Avatarsia, 2022-07-13:
to make the files accessable from the mainsail interface i recommend following changes 

/home/debian/printer_config 
-> 
/home/debian/klipper_config

move into klipper_conf: 
printer.cfg 
mainsail.conf
klipperscreen.conf

add into klipper_conf: 
macro.cfg
mainsail.cfg

add folder:

/home/debian/klipper_logs

add files

/home/debian/klipper_logs/moonraker.log
/home/debian/klipper_logs/klippy.log

add configuration in moonraker.conf
[file_manager]
log_path: /home/debian/klipper_logs/moonraker.log

change services 
Klipper:
ExecStart=/home/debian/klippy-env/bin/python /home/debian/klipper/klippy/klippy.py /home/debian/klipper_config/printer.cfg -l /home/debian/klipper_logs/klippy.log -a /tmp/klippy_uds

Moonraker:
ExecStart=/home/debian/moonraker-env/bin/python /home/debian/moonraker/moonraker/moonraker.py -c /home/debian/klipper_config/moonraker.conf -l /home/debian/klipper_logs/moonraker.log

eliasbakken, 2022-07-24:
Regarding the log locations, what is the benefit to the change? 
/tmp is mounted as tmpfs. 
/ is mounted as ext4
I'm a bit concerned about the long term logging to an eMMC. That's what started killing Teslas in 2019, hehe.

Avatarsia, 2022-07-25:
may we can make a work around and create a syslink to the tmp folder.
there is no 100% benefit out the fact that you can read it any time via the broser in you mashine tab in mainsail

eliasbakken, 2022-07-25:
Ah, so if the logs are located where you suggest (in /home/debian), the logs are available through the Mainsail interface? If so, that is certainly a good benefit. Enough that I think we should try and get it done in this version.

eliasbakken, 2022-07-29:
I think the best solution for this is to crate a folder in /var/log and link to it from /home/debian for both klipper and moonraker. /var/log is type mounted on /dev/zram1, so it seems to be proper ram logging.

eliasbakken, 2022-07-30:
Merged in master.

closed 2022-07-30 by eliasbakken