intelligent-agent/Refactor #325: 3.1.0 - available for Replicape?

mrRobot62, 2022-07-17:
Hi, do you plan, latest refactor-release is available for replicape hardware too?
I would appreciate that :-)

Cheers
Bernd

goeland86, 2022-07-17:
Hi Bernd,

No we're not. The latest developments are primarily klipper enhancements
for recore support, where the replicape has full mainline support already.

If you know of bugs on replicape that we can fix, we're happy to address
them and create new bugfix releases for it though.

Jon

mrRobot62, 2022-07-17:
ok - Jon - thanks for response. Will come back with issues.

BTW:
Today I ruined my eMMC :-O - tried to put SDCard into fstab - now System do not reboot anymore from eMMC :-O 

Any suggestions - without refreshing and reconfiguring everything?

eliasbakken, 2022-07-19:
There are images for BeagleBone you can use that do not automatically flash the eMMC. You can use that to boot the board and then mount the eMMC and edit the fstab file from there.

closed 2022-07-19 by eliasbakken