intelligent-agent/Refactor #333: Add support for system upgrade through Mainsail

eliasbakken, 2022-07-26:
There is now a way to upgrade all necessary packages in the system by running `refactor-apt-upgrade` from the command line. This functionality should be made available in Mainsail. 
The reason why the system can not be upgraded in the normal way is due to the read only filesystem on partition `/boot` and also that a kernel upgrade does not account for the custom device tree binaries in the the boot partition.

goeland86, 2022-11-15:
After running a test on ReFactor 3.1.0-RC2 about what packages to hold back for a system upgrade to work, it seems that running
`apt-mark hold linux-dtb-current-sunxi64 linux-image-current-sunxi64` does the trick just fine and does not mess up boot by overwriting the custom dtbs being added.

eliasbakken, 2023-10-30:
It is now possible to run a system upgrade through mainsail, but it requires manual intervention if the kernel and device trees get updated. A kernel upgrade will overwrite the file `/boot/dtb/allwinner/sun50i-a64-recore.dtb` and replace it with a generic recore device tree binary that works for all versions of recore, but do not prepare the board for klipper. 
A workaround is to manually link sun50i-a64-recore.dtb to sun50i-a64-recore-a7.dtb (or whatever hardware version of the board you have). 
```
sudo mount -o remount,rw /boot
cd /boot/dtb/allwinner/
sudo ln -sf sun50i-a64-recore-a7.dtb sun50i-a64-recore.dtb
sudo reboot
```

eliasbakken, 2023-11-28:
System updates are possible, but they are not enabled by default. Users can still enable system updates from Mainsail by editing the moonraker.conf file. When I ran this recently, KlipperScreen was taken over by `getty@tty1.service`, the plymoth logo was replaced by the standard Armbian logo.