intelligent-agent/Refactor #349: Folder ~/gcode_files mounts ro

shuffleznl, 2022-08-05:
After flashing/dd/writing image from USB, rebooting to emmc with USB still attached, the USB-stick is mounted read-only into the `~/gcode_files/usb` folder. The folder remains ro after removing usb.

eliasbakken, 2022-08-06:
Yes, the `usb` folder is mounted as read-only. This is done on purpose as the intention is to move files from one computer to the printer in case the printer is not connected to the network. 
I think there might be some user right issues if it is mounted RW, but I don't remember exactly right now. Maybe it was done to make the chance of corruption of the drive smaller.

eliasbakken, 2023-04-18:
The lost+found folder can be made readable by the user `chmod +r`. This can be done either as a part of the mounting in refactor or it can be done by Reflash.

eliasbakken, 2023-04-26:
In the latest Refactor (not yet published), I've made the lost+found folder readable by all. This does remove one error message, but is replaced by another where the symlink "dtb" is already watched. I've not found a simple way to move past this. 
Looking at the description of the bug again, I see that there is another problem. That the folder is not removed once the usb drive has been removed. This can probably be solved.