intelligent-agent/Refactor #354: Replace mjpg-streamer with ustreamer

goeland86, 2023-01-26:
It seems like mjpg-streamer is not necessarily the best choice for streaming the webcam.

MainsailOS (the pi image) ships with https://github.com/pikvm/ustreamer installed instead.

It doesn't seem that much more complicated to switch, and the API being compatible with mjpg-streamer's means we benefit.

In addition works with Janus, meaning that Obico (formerly TheSpaghettiDetective) webcam streaming will work as well (incoming PR for that when I've validated the work on my fork).

goeland86, 2023-01-26:
Documentation for ustreamer/janus installation is well-documented here in crowsnest: https://github.com/pikvm/ustreamer/blob/master/docs/h264.md

eliasbakken, 2023-04-28:
Merged in master

closed 2023-04-28 by eliasbakken