intelligent-agent/Refactor #364: Running grep on root with a USB drive inserted causes a panic

eliasbakken, 2023-04-14:
Running `grep -r "test-test" /` for instance causes a complete freeze.

eliasbakken, 2023-04-20:
It seems having the USB drive with Refactor installed causes some instability and seems to also cause a freeze during booting. It might be related to the double mounting in /etc/fstab

goeland86, 2023-04-21:
Hmm. In that case we should really be doing a uuid based fstab, and use a
command street dd to change the uuid of the emmc partitions.
Or the other way around, changing the USB drive's.

eliasbakken, 2023-04-21:
I'm getting an error message ```musb-sunxi 1c19000.usb: Error unknown readw offset 0``` and repeated several times before the freeze.

eliasbakken, 2023-04-24:
Upgrading to kernel 6.1 and running `cat /sys/kernel/debug/usb/musb-hdrc.4.auto/regdump` produces the `Error unknown readw offset` issue. 
It seems that doing anything with `/sys/kernel/debug/psci` can also lead to a crash.

goeland86, 2023-04-24:
Uh... The values in the debug may need to be read by special utils or when
the kernel is booted with debug arguments?

Otherwise, yeah, i can see you could cause crashes with that, especially if
you're trying to access values that are otherwise restricted because of
security features that are enabled by default?

closed 2023-04-26 by eliasbakken