intelligent-agent/Refactor #382: Timer too close

eliasbakken, 2023-09-09:
I've experienced an issue where I get the "timer too close" during probing of the bed on a Voron 2.4. It is not consistent when the error happens, it can happen early or late in the probing process or not at all. The CPU usage on my printer is around 20-30%, mostly due to the transcoding of the webcam data I think (a Logitech C270).

goeland86, 2023-09-10:
I've had that recently on the 3.2.0-RC3, what fixed it was to edit `/etc/init.d/cpufrequtil` and change the governor from `ondemand` to `performance`. It locks the CPU at full speed after a reboot. This means you need a heatsink on the CPU and airflow around it to avoid thermal throttling later on though.
Debating about whether to issue a PR for that, or whether we should experiment with the `userspace` governor and have the user choose what clock speed to use based on the cooling available?

shuffleznl, 2023-09-26:
Happens to me on A5 on 3.1.3 (I think, currently not near machine) too.
Specifically in Arc G3 during fast infill. Repeatable (thrice) to the exact same point in print.

eduncan911, 2023-09-26:
Try disconnecting the webcam on a repeatable print.  Also, too high of microstepping could be an issue (try 16 or 32 for a while).

goeland86, 2023-09-27:
Also if you have a good cooling setup on the main a64 chip, you can try setting the CPU frequency to static high speed instead of dynamic with

`cpufreq-set -c 0 -g performance`

Repeat for CPU 1,2 and 3 by updating the parameter for -c.

Since doing it on a startup script I've not been able to trigger the issue... but the main chip does heat up more as a result.

shuffleznl, 2023-09-28:
> Try disconnecting the webcam on a repeatable print. Also, too high of microstepping could be an issue (try 16 or 32 for a while).

Dont have a webcam connected. Will test other microsteps!

shuffleznl, 2023-09-28:
> Also if you have a good cooling setup on the main a64 chip, you can try setting the CPU frequency to static high speed instead of dynamic with
>
> `cpufreq-set -c 0 -g performance`
>
> Repeat for CPU 1,2 [...]

Have a 120mm fan blowing on the board, should I add heatsink for this?

goeland86, 2023-09-28:
No, if you have a fan that big blowing on it you'll be alright without one.

shuffleznl, 2023-09-30:
Changing govenor sadly doesnt fix issue on A5 with 3.1.3:
[image]

eliasbakken, 2023-10-30:
This has been merged in master. It seems to work on A7, not yet tested on A6. Closing for now, and it can be opened again it necessary.

closed 2023-10-30 by eliasbakken